PowerBroker2/pySerialTransfer #2: Unable to create SerialConnection to /dev/serial0 & /dev/serial1 on raspberry pi

robberwick, 2019-12-27:
Under Raspbian on a raspberry pi, there are `serial0` and `serial1` symlinks created in `/dev` to provide access to the underlying UARTS (see [documentation](https://www.raspberrypi.org/documentation/configuration/uart.md)). As it stands, it is not possible to specify either of these symlinks as the port, because they do not contain the string 'tty' - it gets expanded into `/dev/ttyUSBserial0`

PowerBroker2, 2019-12-28:
hmmm, I'm not very familiar with Linux ports - can you explain each type of port name and maybe a few examples?

robberwick, 2019-12-28:
under linux the ports can have any name. It's not uncommon for usb serial uarts to be `ttyACM0` for instance. What they do have in common though is that they are all block devices so are all in the `/dev/` directory. I think if we test that the string provided is in that format, then we can just use the string as is, and if not, we take the last part after the `/` (if there is one) and append to `/dev/`.

I'll knock together a PR.

PowerBroker2, 2019-12-28:
Fixed in release 1.1.7

PR #4 (2019-12-29): Use serial.tools.list_ports to determine valid values
Resolves #2

Use pySerial tools to determine the list of available serial ports on the system ([docs](https://pyserial.readthedocs.io/en/latest/tools.html)).

Test `port` parameter against full device path and last path fragment, using the os path seperator to split the provided value.

This works for me - it means I can specify my port using shorthand ('serial0', 'ttyUSB0') or whatever, or by full path ('/dev/ttyUSB0'), but those values are tested against those available to pyserial to ensure that they're at least vaguely sane.

I hope thi [... cut]
closed 2019-12-28 by PowerBroker2