PowerBroker2/pySerialTransfer #10: Problem transmitting doubles from Arduino to Python

2447SN, 2020-03-15:
Hi great library, it is very useful, thank you.
I am having trouble sending some data from Arduino to Python. I have it working nicely if my data type is a single byte but if I change to an int or a double the python library isn't behaving as I would expect. Here is an example.
In Arduino I transmit two doubles each with a size of 8 bytes

double var1 = 1;
double var2 = 2;
baseStationSerial.txBuff[0]= var1;
baseStationSerial.txBuff[1]= var2;
baseStationSerial.sendData(16);

In python I read the received bytes into an array and print the array.

response = bytearray()
for index in range(model.link.bytesRead):
       response.append(model.link.rxBuff[index])
print(response)

This is what I get in the array
bytearray(b'\x01\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')

This is what I would expect
bytearray(b'\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00')

Any ideas?

Thanks

PowerBroker2, 2020-03-15:
Can you post your entire Python and Arduino programs along with a description of what your project is supposed to do?

PowerBroker2, 2020-03-15:
Note that the contents of the txBuff should be byte-sized and no larger. If you want to stuff larger values into txBuff, you will have to use a combination of `msb(val)`, `lsb(val)`, and `byte_val(val, pos)` to break down your value's pieces into individual bytes to be then stuffed into the txBuff.

You can also use Python's `struct` package to do byte stuffing as well (as found in the Python application [Thunder Viewer](https://github.com/PowerBroker2/Thunder_Viewer/blob/master/src/Thunder_Viewer.py)):
```Python
    def stuff_float(self, val, start_pos):
        '''
        Description:
        ------------
        Insert a 32-bit floating point value into the (pySerialTransfer) TX
        buffer starting at the specified index
        
        :param val:       float - value to be inserted into TX buffer
        :param start_pos: int   - index of TX buffer where the first byte of
                                  the float is to be stored in
        
        :return start_pos: int - index of the last byte of the float in the TX
                                 buffer + 1
        '''
        
        val_bytes = struct.pack('f', val)
        
        self.transfer.txBuff[start_pos] = val_bytes[0]
        start_pos += 1
        self.transfer.txBuff[start_pos] = val_bytes[1]
        start_pos += 1
        self.transfer.txBuff[start_pos] = val_bytes[2]
        start_pos += 1
        self.transfer.txBuff[start_pos] = val_bytes[3]
        start_pos += 1
        
        return start_pos
```

2447SN, 2020-03-15:
Ok thanks I think that answers my question.

closed 2020-03-15 by 2447SN

PowerBroker2, 2020-03-15:
No problem! Let me know if you still have questions