PowerBroker2/pySerialTransfer #13: Need byte transfer.

Menginventor, 2020-05-05:
I dint see byte type in your code. please implement it.
I have more efficiency when working with byte variable instead of send them by int.

PowerBroker2, 2020-05-05:
If you need to stuff bytes, you can do it manually by accessing the contents of the lists `link.txBuff` and `link.rxBuff`. If your "bytes" are ASCII, you can simply transfer a string using `link.tx_obj()` and `link.rx_obj()`.

Does this help?

PowerBroker2, 2020-05-06:
Also, you might be able to send a byte array using `link.tx_obj()` and `link.rx_obj()`.

Menginventor, 2020-05-09:
Actually, I modify you library to my version.
Did you implement 2 byte int for arduino? I see you use 4 byte int in python program

PowerBroker2, 2020-05-09:
Because Python's int size is 32-bit, integers handled by `link.tx_obj()` and `link.rx_obj()` will be 32-bit. If you really need to send 16-bit ints, you can use `byte_val(val, pos)` to find the two lowest bytes of `val` and stuff them (little-endian) manually into `link.txBuff'.

Menginventor, 2020-05-11:
Seem like communicate between Arduino and python isn't your primary propose. It okay I totally understand. thank for your clean code I can find where I need to modify for more  convenient.

closed 2020-05-11 by Menginventor

PowerBroker2, 2020-05-11:
> Seem like communicate between Arduino and python isn't your primary propose.

Python <--> Arduino communication **was** the entire purpose of creating this Python package...