PowerBroker2/pySerialTransfer #14: Only read when a new sample is ready?

crossband, 2020-05-07:
Hi PB2,

I'm using your awesome library to read sensors connected to the serial port and save the data to an HDF5 file. My problem is that I cannot prevent the code "reading of non-existing samples" and writing zeros. The slower the sample rate, the more zeros it writes. I need to implement an instruction which checks if a new sample is ready.

Would you please tell how could I do it?

Best regards,
Adam

``` Python
import struct
import h5py
import pandas as pd
import numpy as np
from pySerialTransfer import pySerialTransfer as txfer
import time

buffLen = 10
hdfFile = 'dtype.hdf5'
dataTypes = ['<i2', '<f4', '<f4', '<f4']
dataSets = ['r', 'x', 'y', 'z']

if __name__ == '__main__':
    try:
        # Define the serial connection
        link = txfer.SerialTransfer('COM3', 460800)
        # Connect to port and check
        if link.open():
            # Wait serial to start
            time.sleep(3)
            # Create an HDF5 file
            with h5py.File(hdfFile, 'w') as hf:
                # Create HDF5 datasets for all sensors
                for i in range(len(dataSets)):
                    hf.create_dataset(dataSets[i], (0,), dtype=dataTypes[i], chunks=True, compression="gzip", maxshape=(None,))

                # Write 500 batches of data
                for i in range(500):
                    # Batch buffer
                    buffer = []
                    # Check if buffer empty
                    if not buffer:
                        # Fill the buffer with lists of values
                        for count in range(buffLen):
                            dataList = [0 for x in range(4)]
                            if link.available():
                                dataList[0] = struct.unpack('h', bytes(link.rxBuff[0:2]))[0]
                                dataList[1] = struct.unpack('f', bytes(link.rxBuff[2:6]))[0]
                                dataList[2] = struct.unpack('f', bytes(link.rxBuff[6:10]))[0]
                                dataList[3] = struct.unpack('f', bytes(link.rxBuff[10:14]))[0]
                            elif link.status < 0:
                                print('ERROR: {}'.format(link.status))
                            buffer.append(dataList)
                    # Pandas data frame out of buffer
                    df = pd.DataFrame(buffer, columns=dataSets)
                    #print(df)
                    # Write buffer data frame to HDF5 file
                    for colName in df.columns:
                        hf[colName].resize((hf[colName].shape[0] + len(df.index),))
                        hf[colName][-len(df.index):] = df.loc[:, colName]

    except KeyboardInterrupt:
        link.close()

```

[image]

crossband, 2020-05-07:
I made a fatal mistake here and did not consider that `for count in range(buffLen):` will continue to counting even when there is no ne sample available. Correct code will be like this:
```python
import struct
import h5py
import pandas as pd
import numpy as np
from pySerialTransfer import pySerialTransfer as txfer
import time

buffLen = 10 # Length of single batch
hdfFile = 'dtype.hdf5'
dataTypes = ['<i2', '<f4', '<f4', '<f4']
dataSets = ['r', 'x', 'y', 'z']

if __name__ == '__main__':
    try:
        # Define the serial connection
        link = txfer.SerialTransfer('COM3', 460800)
        # Connect to port and check
        if link.open():
            # Wait serial to start
            time.sleep(3)
            # Create an HDF5 file
            with h5py.File(hdfFile, 'w') as hf:
                # Create HDF5 datasets for all sensors
                for i in range(len(dataSets)):
                    hf.create_dataset(dataSets[i], (0,), dtype=dataTypes[i], chunks=True, compression="gzip", maxshape=(None,))

                # Write N batches of data
                for i in range(500):
                    # Batch buffer
                    buffer = []
                    # Fill the buffer with lists of values
                    for count in range(buffLen):
                        dataList = [0 for x in range(4)]
                        while len(buffer) < buffLen:
                            if link.available():
                                dataList[0] = struct.unpack('h', bytes(link.rxBuff[0:2]))[0]
                                dataList[1] = struct.unpack('f', bytes(link.rxBuff[2:6]))[0]
                                dataList[2] = struct.unpack('f', bytes(link.rxBuff[6:10]))[0]
                                dataList[3] = struct.unpack('f', bytes(link.rxBuff[10:14]))[0]
                                buffer.append(dataList)
                            elif link.status < 0:
                                print('ERROR: {}'.format(link.status))
                            else:
                                continue

                    # Pandas data frame out of buffer
                    df = pd.DataFrame(buffer, columns=dataSets)
                    #print(df)
                    # Write buffer data frame to HDF5 file
                    for colName in df.columns:
                        hf[colName].resize((hf[colName].shape[0] + len(df.index),))
                        hf[colName][-len(df.index):] = df.loc[:, colName]

    except KeyboardInterrupt:
        link.close()
    except:
        import traceback
        traceback.print_exc()

        link.close()
```

PowerBroker2, 2020-05-07:
No prob, glad it's working for you!

crossband, 2020-05-07:
Hrrmm. Now I am facing another strange problem. `for count in range(buffLen):` keeps appending the same `dataList` to the `buffer`. The Arduino board generates random numbers, so it is not the data source.
Here is the `buffer` cycles printed:

[image]

Arduino code if needed:
```C++
#include <Arduino.h>

#include "SPI.h" // necessary library
#include "SerialTransfer.h"

// Serial connection library
SerialTransfer myTransfer;

int16_t Result_Rate;
float Result_Acc_X;
float Result_Acc_Y;
float Result_Acc_Z;

volatile bool interruptCounter;
// Timer interrupt
hw_timer_t *timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;
void IRAM_ATTR onTimer(){
    portENTER_CRITICAL_ISR(&timerMux);
    interruptCounter = true;
    portEXIT_CRITICAL_ISR(&timerMux);
}

// Send data over serial port
void sendToPC(int16_t data1, float data2, float data3, float data4){
    uint8_t sendLen = 0;

    myTransfer.txObj(data1, sizeof(data1), sendLen);
    sendLen += sizeof(data1);

    myTransfer.txObj(data2, sizeof(data2), sendLen);
    sendLen += sizeof(data2);

    myTransfer.txObj(data3, sizeof(data3), sendLen);
    sendLen += sizeof(data3);

    myTransfer.txObj(data4, sizeof(data4), sendLen);
    sendLen += sizeof(data4);

    myTransfer.sendData(sendLen);
}

void setup(){
    setCpuFrequencyMhz(80); //Set CPU clock
    // Timer interrupt
    timer = timerBegin(0, 80, true); // prescaler
    timerAttachInterrupt(timer, &onTimer, true);
    timerAlarmWrite(timer, 3333, true); // @80Mhz /(80 * 435) = 2.3kHz
    timerAlarmEnable(timer);

    Serial.begin(460800);

    // Start serial library
    myTransfer.begin(Serial);
}

void loop(){
    if (interruptCounter)    {
        // Timer interrupt
        portENTER_CRITICAL(&timerMux); // Enter to critical sectionn
        interruptCounter = false;
        portEXIT_CRITICAL(&timerMux); // Exit from critical sectionn

        // Convert data
        Result_Rate = random(-50, 50);
        Result_Acc_X = random(-2, 2);
        Result_Acc_Y = random(-2, 2);
        Result_Acc_Z = random(-2, 2) ;

        sendToPC(Result_Rate, Result_Acc_X, Result_Acc_Y, Result_Acc_Z);
    }
}
```

PowerBroker2, 2020-05-07:
I've taken a look at both the Python and Arduino programs, but nothing looks bugged. I also double-checked the Python parsing code and everything checks out.

Can you do prints of `dataList` and `buffer` as you get new data?

A few notes:

- Update both Python and Arduino versions of the lib to the most current version
- pySerialTransfer now has built-in functions to automatically parse multi-byte objects out of the payload buffer using `link.rx_obj()` (the example in the readme shows how to use it)
- `random()` produces an int, not a float
- It would be easiest to create a float array and simply call `myTransfer.txObj(floatArray, sizeof(floatArray));` without having to use `sendToPC()` at all. You could then use `link.rx_obj()` to parse the float list and directly insert it into your buffer.

crossband, 2020-05-11:
If I run bare pySerialTransfer code to print each sample separately:
```python
import struct
from pySerialTransfer import pySerialTransfer as txfer

if __name__ == '__main__':
    try:
        link = txfer.SerialTransfer('COM3', 460800)
        link.open()
        while True:
            dataList = [0 for x in range(4)]
            if link.available():
                dataList[0] = struct.unpack('h', bytes(link.rxBuff[0:2]))[0]
                dataList[1] = struct.unpack('f', bytes(link.rxBuff[2:6]))[0]
                dataList[2] = struct.unpack('f', bytes(link.rxBuff[6:10]))[0]
                dataList[3] = struct.unpack('f', bytes(link.rxBuff[10:14]))[0]
                print(dataList)
            elif link.status < 0:
                print('ERROR: {}'.format(link.status))

    except KeyboardInterrupt:
        link.close()
```
Everything works well and no repeating values:

`[20619, -1.36897611618042, 0.5896615982055664, -0.6402678489685059]
[-2670, -1.057100772857666, 1.3561162948608398, 0.2653048038482666]
[31142, -0.46735239028930664, 0.9030382633209229, 0.12019228935241699]
[-10340, 1.3646280765533447, -1.8809614181518555, -0.8457545042037964]
[-19135, 1.157583236694336, -1.395113468170166, -1.4061834812164307]
[17235, -1.046063780784607, -0.4761742353439331, -1.0625197887420654]
[-7702, 1.560126781463623, 0.011037826538085938, 1.4518990516662598]
[-17249, -0.4047003984451294, 1.134134292602539, 0.37402963638305664]
[-3774, -1.633435845375061, 0.1748976707458496, -0.49981367588043213]
[20623, 0.9814577102661133, -1.3584123849868774, 0.4584476947784424]
[-6709, 0.14779186248779297, -1.5563654899597168, -0.781969428062439]
[24191, -1.8885451555252075, 0.5318546295166016, 0.6611771583557129]
[-15853, -1.7237610816955566, 1.8906402587890625, 1.8167715072631836]
[28670, 0.8412449359893799, 0.516033411026001, -1.0724363327026367]
[-14909, 1.0932157039642334, -1.083557367324829, 1.4916131496429443]
[31377, -1.4191505908966064, 0.3694190979003906, 0.891836404800415]
[30739, -0.5835210084915161, 1.9831397533416748, 0.5637550354003906]
[-19205, 0.4315969944000244, -0.3251723051071167, 0.4040651321411133]
[14780, 0.8138589859008789, 1.146920919418335, -1.2539292573928833]
[-28505, 0.8262960910797119, 0.22207164764404297, 1.9753494262695312]
[27464, 0.23137617111206055, 1.6449220180511475, -1.8255515098571777]`

If I print both `dataList` and `buffer` values simultaneously:
```python
for count in range(buffLen):
    dataList = [0 for x in range(4)]
    while len(buffer) < buffLen:
        if link.available():
            dataList[0] = struct.unpack('h', bytes(link.rxBuff[0:2]))[0]
            dataList[1] = struct.unpack('f', bytes(link.rxBuff[2:6]))[0]
            dataList[2] = struct.unpack('f', bytes(link.rxBuff[6:10]))[0]
            dataList[3] = struct.unpack('f', bytes(link.rxBuff[10:14]))[0]
        elif link.status < 0:
            print('ERROR: {}'.format(link.status))
        else:
            continue
        buffer.append(dataList)
        print(dataList)
        print(buffer)
```
I can see my problem, values are repeating. Every `while len(buffer) < buffLen:` cycle overwrites previous values.
But what is the most interesting, despite I generate random numbers, running my `buffer`ing code results exactly the same values every time.

[image]

### So the problem is not related to the library.

Although the libraries were working fine, I have updated them.  
Thanks again PB2!
P.S. I will later try `txObj()` and `rx_obj()`.

closed 2020-05-11 by crossband