PowerBroker2/pySerialTransfer #17: CPU load optimization of the time critical infinite loop.

crossband, 2020-06-15:
I use the following Python script to record data at 500 Hz on Raspberry Pi. Since it constantly checks for samples in the loop, it overloads the CPU almost 100%.
I would like to ask if it could be done differently by reducing CPU usage.
Thanks!

```python
#!/usr/bin/env python3
from pySerialTransfer import pySerialTransfer as txfer
import os
import time

# get file directory
__location__ = os.path.realpath(os.path.join(os.getcwd(), os.path.dirname(__file__)))

buffLen = 2 ** 12                                                   # buffer size
maxFileSize = 100 * 10 ** 6                                         # maximum file size
print('Max file size: ' + str(maxFileSize/1000) + ' MB')

if __name__ == '__main__':
    try:
        link = txfer.SerialTransfer('/dev/ttyUSB0', 460800)
        if link.open():
            print("Connected")
        # Wait serial to reset
        time.sleep(3)

        while True:
            # create a new file and open
            fileName = str(round(time.time())) + 'cms.bin'
            filePath = os.path.join(__location__, fileName)
            file = open(filePath, 'wb')
            print('New file: ' + fileName)
            with open(fileName, 'wb') as file:
            # start writing data to new file
                while True:
                    rawBuffer = bytearray()
                    for count in range(buffLen):
                        # write only if new data available
                        if link.available():
                            rawBuffer.extend(bytes(link.rxBuff[0:11]))
                        elif link.status < 0:
                            print('ERROR: {}'.format(link.status))
                        else:
                            continue
                    file.write(rawBuffer)
                    # break to create a new file
                    if os.path.getsize(fileName) > maxFileSize:
                        break

    except KeyboardInterrupt:
        link.close()
```

PowerBroker2, 2020-06-15:
Please use issues for library problems - this is a generic coding style/design question.

However, you can probably reduce CPU load by adding `time.sleep(0.002)` call before `if link.available():`

closed 2020-06-15 by PowerBroker2

crossband, 2020-06-16:
Thanks PB, yes you are right. But if I'm not mistaken, you originally wrote this library for the use in robotics, I thought there was already a method to get messages iteratively.