PowerBroker2/pySerialTransfer #18: Receiving from Arduino works but sending to Arduino doesn't

BrainStone, 2020-07-01:
Hi there.  
So I'll be frank, I'm not 100% sure it's the Python library but it looks like it is.

So the Python library can receive data without issues. But sending the data doesn't work.  
From my testing when sending a single byte the arduino received 5 bytes (``I checked Serial.available()``) instead of the 7 I would expect when sending a single payload byte and cnsequently the ``SerialTransfer::available`` call on the Arduino always returns 0. (That's what's making me suspect the Python library)  
Having the library in debug mode didn't throw any errors so I'm guessing it doesn't properly send the start byte in the first place.

I tested it with both Python 3.7.3 and Python 2.7.16, both with the same result.  
The testing was conducted on a Debian 10 machine and the Arduino was connected over an internal USB-Port.

PowerBroker2, 2020-07-03:
Thank you for bringing this to my attention. I forgot to add the "packet ID" byte field in the packet transmitted from Python to the Arduino. This will be fixed, pushed, and published ASAP.

However, I'm not sure why you're getting 2 less bytes than intended instead of just 1. I'll let you know when this gets fixed so you can update both pySerialTransfer and SerialTransfer.h.

PowerBroker2, 2020-07-03:
Issue fixed in [2.0.1](https://github.com/PowerBroker2/pySerialTransfer/releases/tag/2.0.1) - make sure to update your libs

closed 2020-07-03 by PowerBroker2

BrainStone, 2020-07-03:
That fixed it. Not sure why it showed 5 bytes when it in reality was 6 but in any case it's working properly now. Thanks for the fast fix.