PowerBroker2/pySerialTransfer #21: Error in example script

McAroni, 2020-07-16:
Hey there, 
the given example script throws this error:

```
"D:\Programme\Portable Python 2.7.6.1\App\PyCharm\..\pythonw.exe" "D:/Python Skripte/Screen2Bulb/serialExample.py"
Traceback (most recent call last):
  File "D:/Python Skripte/Screen2Bulb/serialExample.py", line 7, in <module>
    link = txfer.SerialTransfer('COM17')
  File "D:\Programme\Portable Python 2.7.6.1\App\lib\site-packages\pySerialTransfer\pySerialTransfer.py", line 125, in __init__
    for p in serial_ports():
  File "D:\Programme\Portable Python 2.7.6.1\App\lib\site-packages\pySerialTransfer\pySerialTransfer.py", line 94, in serial_ports
    return [p.device for p in serial.tools.list_ports.comports(include_links=True)]
TypeError: comports() takes no arguments (1 given)

Process finished with exit code 0
```

I just downloaded the zip from github and copied this example.
Changing the COM-Port to a correct one doesn't change anything. 

I don't think I made a mistake, did I?

PowerBroker2, 2020-07-16:
According to [the documentation](https://pyserial.readthedocs.io/en/latest/tools.html), there is an optional argument that matches the one provided in the line of code that threw the error:
```python
serial.tools.list_ports.comports(include_links=False)
```
I've never had this error thrown when testing/using the library and after reviewing the documentation, I don't see how your error is possible or even valid. 

**I do see you're using Python 2.** That may be the problem - update to Python 3 and reinstall the necessary libraries and see if things get fixed.

McAroni, 2020-07-17:
Thanks for the answer - yes, this was the problem.

closed 2020-07-17 by PowerBroker2