PowerBroker2/pySerialTransfer #23: something weird in "find_last" function

Menginventor, 2020-08-09:
According to your code
`  def find_last(self, pay_len):
        '''
        Description:
        ------------
        Finds last instance of the value START_BYTE within the given
        packet array

        :param pay_len: int - number of bytes in the payload

        :return: int - location of the last instance of the value START_BYTE
                       within the given packet array
        '''

        if pay_len <= MAX_PACKET_SIZE:
            for i in range(pay_len - 1, 0, -1):
                if self.txBuff[i] == START_BYTE:
                    return i
        return -1`
This function suppost to scan all data but  range(pay_len - 1, 0, -1) just stop before reach index 0.
This is on purpose or not?

PowerBroker2, 2020-08-09:
I can't understand what you're trying to say. Can you explain your question?

Menginventor, 2020-08-09:
Yes. So you need to replace every byte that equals to START_BYTE so you scan whole packet to see all byte but some how you prefre to see the last byte first.
If we got 5 bytes in our packet, we need to scan all index from 0 to 4. For each loop, index value should be
```
4
3
2
1
0
```
but your code .   "for i in range(pay_len - 1, 0, -1):"
will produce
```
4
3
2
1
```
It not reaching zero, the first index of packet. At this point I wonder this is on purpose or not?

PowerBroker2, 2020-08-10:
Ah, I see. I vaguely remember writing that function but don't remember any specific reason why I skipped the first index. I think I need to change it to:

```Python
  def find_last(self, pay_len):
        '''
        Description:
        ------------
        Finds last instance of the value START_BYTE within the given
        packet array

        :param pay_len: int - number of bytes in the payload

        :return: int - location of the last instance of the value START_BYTE
                       within the given packet array
        '''

        if pay_len <= MAX_PACKET_SIZE:
            for i in range(pay_len - 1, -1, -1):
                if self.txBuff[i] == START_BYTE:
                    return i
        return -1
```

Menginventor, 2020-08-10:
I already change that part. It work fine with the rest of code. (I've test with serialTransfer on Arduino).
I really admire your work trust it's make my life a lot easier.
I found this piece of code while I try to modify your work to make easier version.
This my work [https://github.com/Menginventor/SerialCom](url) in case you might want to take a quick look.

closed 2020-08-10 by Menginventor