PowerBroker2/pySerialTransfer #25: Portability vs. non-adjustable endianness in Python implementation

martincizek, 2020-09-07:
To ensure portability, SerialTransfer can either
1. use network byte order on all implementations or
2. let the user choose on all its implementations.

PowerBroker2/SerialTransfer API works with low-level integer types and uses native host byte order. This is OK, as the users in C++ have well-known ways how to ensure endianness compatibility on these low-level integer types, should they need it.

PowerBroker2/pySerialTransfer API works with high-level Python int type and uses struct packing in Python-host-native byte order. This might be an issue, as changing endianness on top of Python int type rapes its semantics and requires a hacky solution.

I'd suggest adding an optional parameter to `tx_obj()` and `rx_obj()` methods, allowing the users to pass byte order to the calls. The value would be according to https://docs.python.org/3/library/struct.html#byte-order-size-and-alignment, native by default.

Does it make sense?

PowerBroker2, 2020-09-08:
Yes, this makes sense to add once I can get around to it - thanks!

martincizek, 2020-09-08:
> Yes, this makes sense to add once I can get around to it - thanks!

You're welcome! I've created a PR implementing it, hope it helps. :)

PR #26 (2020-09-08): Add byte_format parameter to rx_obj() and tx_obj() methods
 Resolves #25.
commit 39f53fc6 (2020-09-08): Add byte_format parameter to rx_obj() and tx_obj() methods. Resolves #25.
closed 2020-09-08 by PowerBroker2