PowerBroker2/pySerialTransfer #27: ValueError: bytes length not a multiple of item size

gordon13, 2020-10-18:
I'm trying to receive an array of uint16_t from the Arduino.  

This is how I send from the Arduino:
```
    uint16_t sendSize = 0;
    uint16_t header[3] = {1, 1, 6};
    sendSize = myTransfer.txObj(header, sendSize);
    myTransfer.sendData(sendSize);
```

This is what I used to received on the Python side:
```
        header = link.rx_obj(obj_type=list,
                    obj_byte_size=3*2,
                    list_format="i")
```

And this is the error I get:
```
 File "C:\ProgramData\Miniconda3\lib\site-packages\pySerialTransfer\pySerialTransfer.py", line 287, in rx_obj
    arr = array(list_format, buff)
ValueError: bytes length not a multiple of item size
```

I tried debugging it by adding print statements in the library, and it fails at this point in pySerialTransfer.py:
```
            elif list_format:
                arr = array(list_format, buff)
                return arr.tolist()
```

This is the content of the buff variable when the data is received from the Arduino: 
b'\x01\x00\x01\x00\x06\x00'

If I change the "list_format" variable to 'b', then it works, however it interprets th \x00 as individual 0s instead of combining two bytes together. This is what I get:
```
>>> array('b', b'\x01\x00\x01\x00\x06\x00')
array('b', [1, 0, 1, 0, 6, 0])
```

I would expect this:
```
>>> array('i', b'\x01\x00\x01\x00\x06\x00')
array('b', [1,  1, 6])
```

This is happening on Python 3.7 64 bit
Any ideas?

PowerBroker2, 2020-10-18:
This is because your Python expects integers to be 32 bytes long. You can either change the Arduino code to send 32 bit ints or change your Python to the following:

```Python
        header = link.rx_obj(obj_type=list,
                                        obj_byte_size=3*2,
                                        list_format="h")
```

Let me know if this fixes things

gordon13, 2020-10-19:
> This is because your Python expects integers to be 32 bytes long. You can either change the Arduino code to send 32 bit ints or change your Python to the following:
>
> ```python
> header = [...]

Ah this solved it! I think I was just confusing ints and shorts and their respective sizes. I missed the "minimum" part in the Array documentation that described the "list_format". This makes sense now.

closed 2020-10-20 by PowerBroker2