PowerBroker2/pySerialTransfer #30: How to receive list of integers from Python to Arduino?

natchapol2347, 2020-11-25:
I'm trying to send a list of int and boolean using txObj() in python and receive in Arduino using rxObj(). 

Here's my python test code snippet: 
```
from pySerialTransfer import pySerialTransfer as txfer
link=txfer.SerialTransfer('COM6',9600)
link.open()
values=[128,128,128,128,False,False,False,False]
list_=link.tx_obj(values)
link.send(list_)

```
And here's my arduino code snippet:
```
struct data_package
{
  uint32_t go_x; //Right and left control
  uint32_t go_y ; //Forward and backward control
  uint32_t turn_x ; //Camera Control right and left
  uint32_t turn_y ; //Camera Control up and down
  bool square;
  bool x;
  bool o;
  bool triangle;   
};
data_package data;
void checkEvent()
{
    if(myTransfer.available())
  {
    myTransfer.rxObj(data, sizeof(data));
    lastReceiveTime = millis();

 
  }

}
```
It seems like the value I received is not consistent on the Arduino, perhaps because the integer bit size of Python is not the same as C in Arduino? How can I fix it? I am also using Arduino Uno with one port, so I cannot really debug it for now.

PowerBroker2, 2020-11-27:
What sort of values are you receiving? Can you give examples of inaccuracies and inconsistencies? Your test code looks good - are you using Python3 and the latest version of the libraries?

natchapol2347, 2020-12-09:
Sorry for the late reply, I did not look more into it after posting this. Instead I just used the normal `while(Serial.available()>=8) ` for my 8 indexes. But what happened was that it looks like only the first received index of the list is read and ignores the others. 

Anyhow it would be nice if someone can add to this as I might use the library in the future. It looks very neat!

closed 2020-12-09 by PowerBroker2