PowerBroker2/pySerialTransfer #33: Encoding Many Variables Correctly

jmdelahanty, 2021-04-22:
Hello again PowerBroker!

The lab is happy with how the Serial Transfer works and has asked me to try and implement a transfer of all experimental variables to the Arduino. I'm wondering how I should best transfer many different variables using the package. Should I send things using individual functions one at a time or should I open communications once and send it all during the same transmission? Do you think it matters which way I do it?

PowerBroker2, 2021-04-22:
Hey Jeremy!

Good question - I would say it depends on the total amount of information you want to send. If I were you, I would create [a struct](https://www.geeksforgeeks.org/structures-in-cpp/) that has all of the experiment variables you need. Also be sure to use as small of variable types as possible (i.e. if you know a specific variable will be in the range of 0-255, use a `uint8_t` instead of an `int` or `float`) and also use the ["packed" variable attribute](https://www.keil.com/support/man/docs/armcc/armcc_chr1359124982232.htm). Here's an example from one of my own projects on how to do all that:

```C++
struct __attribute__((__packed__)) telem_struct
{
	uint16_t altitude;      //cm
	float courseAngleIMU;   //degrees
	float rollAngle;        //degrees (positive --> roll right)
	float pitchAngle;       //degrees (positive --> nose up)
	uint16_t pitotPressure; //unitless             MSB                                                                        LSB
	uint8_t validFlags;     //bit encoded bools (reserved, reserved, imuRecent, imuValid, fixRecent, fixValid, linkValid, manualControl)
	float latitude;         //dd
	float longitude;        //dd
	uint16_t UTC_year;      //y
	uint8_t UTC_month;      //M
	uint8_t UTC_day;        //d
	uint8_t UTC_hour;       //h
	uint8_t UTC_minute;     //m
	float UTC_second;       //s
	float speedOverGround;  //knots
	float courseOverGround; //degrees
} telem;
```

Lastly, you need to determine the total size of the struct in bytes. This can be done with: `Serial.println(sizeof(<<struct name>>));`. If that number less than or equal to 254, you can send the whole struct in one go. If the number is greater than 254, you will need to break the transmission up into 2 or more goes.

jmdelahanty, 2021-04-22:
Hello! Thanks again for your help!

So in trying to get the size of the structure, I'm running into an error about using a variable referenced in the structure. Here's what I have so far:
```C++
// Use package SerialTransfer.h from PowerBroker2 https://github.com/PowerBroker2/SerialTransfer
#include "SerialTransfer.h"

// Rename SerialTransfer to myTransfer
SerialTransfer myTransfer;

struct __attribute__((__packed__)) variables_struct {
  const int8_t totalNumberOfTrials; // total number of trials for experiment
  int32_t trialArray[totalNumberOfTrials]; // create trial array
  int32_t ITIArray[totalNumberOfTrials]; // create ITI Array
  int16_t noiseDuration; // length of tone played by speaker
  int16_t punishTone; // airpuff frequency tone in Hz
  int16_t rewardTone; // sucrose frequency tone in Hz
  int8_t USDeliveryTime_Sucrose; // amount of time to open sucrose solenoid
  int8_t USDeliveryTime_Air; // amount of time to open air solenoid
  int16_t USConsumptionTime_Sucrose; // amount of time to wait for sucrose consumption
}
```
From what I read online, I need to use a ```const int``` to reference the a variable. but even after doing so I get this error:
```invalid use of non-static data member 'variables_struct::totalNumberOfTrials'```

Is there a different way to declare a constant integer that I'm missing? Or can I not use a variable in a ```struct``` when it's been declared in the same ```struct```?

PowerBroker2, 2021-04-22:
You can't do dynamic array allocation in a struct like how you're trying to do. That combined with the sheer amount of data leads me to think you should send one packet per trial to be performed. On the receiving side, you could use an array for each setting that holds values for each experiment. The arrays could be dynamically allocated (possible, but not recommended) or have a max number of experiments and make the array lengths static.

Does that make sense?

Also, for the Python side, you'll need to stuff the library's TX buffer with each value individually while keeping close track of the byte size of each value. In order for all the bytes to line up, you'll likely need to use the `val_type_override` argument in the `tx_obj()` member function like this:

```Python
link.tx_obj(USDeliveryTime_Air, start_pos=16, byte_format='', val_type_override='h'):
```

This is because Python ints default to 32-bit, so if you're trying to transfer an 8-bit or 16-bit number, you'll have to specify `b` or `h` (respectively) to get the byte size right. For more info on valid `val_type_override` values, check [this link](https://docs.python.org/3/library/struct.html#format-characters).

jmdelahanty, 2021-04-22:
I think I understand but let me try to make sure. 

My initial idea was to have everything for a particular experiment on the Arduino so I only had to communicate with Python once. In other words, I would send the trial types and relevant time variables before anything got going. Once the trial set runs out, the experiment would be over and the Arduino would just sit and wait for a new set of trials to arrive when Python initiates a new experiment. I didn't mean for the arrays to be changeable during a particular experiment, but I did want the user to be able to adjust the trial count if they wanted to make it longer or shorter on the next experiment.

I think what I understand here is that I should send individual trials instead of the whole array to the Arduino. Is that right? That way I don't have to worry about storing arrays and sending multiple packets?

And thank you pointing that out to me again! I remember from the last question I had that you taught me Python defaults to 32-bit ints. I didn't know how to handle different sizes so thank you for that!

PowerBroker2, 2021-04-22:
You can still provide all the data for all the experiments upfront as you want. Here's how I would do it:

### Data Structures:

```C++
const int MAX_NUM_TRIALS = 100; // make sure this is larger than you will ever need, but smaller than the max value of metadata.totalNumberOfTrials

struct __attribute__((__packed__)) metadata_struct {
  uint8_t  totalNumberOfTrials;       // total number of trials for experiment
  uint32_t trialNumber;               // create trial array
  uint16_t noiseDuration;             // length of tone played by speaker
  uint16_t punishTone;                // airpuff frequency tone in Hz
  uint16_t rewardTone;                // sucrose frequency tone in Hz
  uint8_t  USDeliveryTime_Sucrose;    // amount of time to open sucrose solenoid
  uint8_t  USDeliveryTime_Air;        // amount of time to open air solenoid
  uint16_t USConsumptionTime_Sucrose; // amount of time to wait for sucrose consumption
} metadata;

int32_t trialArray[MAX_NUM_TRIALS]; // create trial array
int32_t ITIArray[MAX_NUM_TRIALS];   // create ITI Array
```

## Sending Data:

1. Send the `metadata` struct with a **packet ID of 0** (check the APIs of both libs on how to set/access this)
2. Send one packet per trial with the following data and a **packet ID of 1**: trialNumber, trialArray[trialNumber], and ITIArray[trialNumber]

## Receiving Data:

Depending on the packet ID, you will parse and save the data to their respective object as defined in the "Data Structures" section.

## Note:

Due to how much data you're trying to move, this will be difficult to get working initially, but once you get the hang of it and work out all the kinks, it will be very scalable and (hopefully) maintainable.

jmdelahanty, 2021-04-23:
Thanks! I will give this a go and see what happens. One thing I'm immediately unsure about is why send a ```trialNumber``` to/from the Arduino? I might be getting confused by this:

In my understanding for what I intend ```trialNumber``` to be, it is an individual element of the ```trialArray```.  With the code here, before each trial, I would send the current ```trialNumber``` to the Arduino and, once that trial is over, send back the ```trialNumber``` just completed as well as the ```trialArray``` and ```ITIArray```. So does this mean that we'll be building the ```trialArray``` and ```ITIArray``` as the experiment moves along? Or am I misunderstanding something?

PowerBroker2, 2021-04-23:
No, you send all the data before you start any experiments. Because we need to split the data across multiple packets, we need a way to interpret each piece of data we receive as we receive it.

Think of it as building each array by sending the index plus the data to be inserted into the array(s). Once the arrays are "built", you can start the experiments/trials.

Does that help?

jmdelahanty, 2021-04-23:
That totally makes sense. I'll give this a try and let you know how it goes! 

Thanks PowerBroker!

jmdelahanty, 2021-04-24:
Okay, so today I've been looking over how to construct the packets and provide an ID for them in Python. Apologies for the butchering I'm about to write to see if I understand how to use it and for the several questions that I'm about to ask.

The ```SerialTransfer``` class allows us to provide an object of some kind and the transmission/receiving buffers created are the max packet size of 244 (from Hexadecimal notation I've learned, 0xFE, as defined in the Class). The class then sets the ```idByte``` position as in the first location of the packet. I _think_ the packet still has no ID set at this point.

In this case, I'll be using the ```send``` function which allows me to set a particular ```packet_id``` for the transmission. Send then initializes an empty ```stack``` and prevents a message longer than the max packet size to be put into the stack. I'm not sure if this means that if I try to fit more than a packet can hold into one transmission, the package knows where it left off and introduces a new start byte for a subsequent packet or if I have to identify where to start next.

The packet is then constructed depending on the type of information contained in the message and written over the connection to the Arduino. 

One thing that I've considered trying is to generate two "payloads" that have a ```sizeof``` function return a value less than 254 and then setting each payload up with it's own ```packet_id```. Is that something that might be helpful or does the package do this allocation for you through some kind of magic?

PowerBroker2, 2021-04-24:
> I'm not sure if this means that if I try to fit more than a packet can hold into one transmission, the package knows where it left off and introduces a new start byte for a subsequent packet or if I have to identify where to start next.

The package doesn't know where it left off, so you're going to have to keep track of that yourself. If you use my suggested method, you won't need to do, though.

> One thing that I've considered trying is to generate two "payloads" that have a sizeof function return a value less than 254 and then setting each payload up with it's own packet_id.

Yeah, I think that would work. The cool thing about software design and engineering in general is that there are no wrong answers as long as they fulfil your requirements

> Is that something that might be helpful or does the package do this allocation for you through some kind of magic?

I'm not sure I understand. What do you mean?

jmdelahanty, 2021-04-24:
> The package doesn't know where it left off, so you're going to have to keep track of that yourself. If you use my suggested method, you won't need to.
> Is that something that might be helpful or does the package do this allocation for you through some kind of magic?

The first quote is what I was trying to reference, I wasn't sure if it somehow knew how to keep track of different packets automatically. Silly attempt at some humor, sorry.

>Yeah, I think that would work. The cool thing about software design and engineering in general is that there are no wrong answers as long as they fulfil your requirements

Awesome! Then I'll try and do this and make it a little easier to track each packet.

Was I correct with using ```send``` to set a packet_id by the way? I wasn't sure if that's the proper way to do it. And then for sending a struct in Python, do I need to use the struct package in conjunction with pySerialTransfer? Since the examples include lists, floats, and strings I wasn't sure if I needed to package the variables in a different way. It's possible I missed it in the documentation...

PowerBroker2, 2021-04-24:
> Was I correct with using send to set a packet_id by the way? 

Yup, that's how you do it

> And then for sending a struct in Python, do I need to use the struct package in conjunction with pySerialTransfer?

Python is a little more difficult when it comes to sending data than on an Arduino. Instead of using the struct package, you'll probably be better off individually stuffing the values in the TX buffer.

One more note: Another idea for getting around this max byte cap is to send the metadada struct and each array as 3 separate packets with unique packet IDs. Super quick and easy

jmdelahanty, 2021-04-27:
Alright, hello again! So over the weekend and today I've gotten to try and implement multiple packets in Python but I'm really unsure if I've done this correctly. I've tried to generate each array and packet in a function so they can be written to a config file and to the Arduino respectively. Here's one of the functions just for some of the metadata:
```Python
#### Experiment Metadata Function ####
# TODO: Either give a config file that contains these values or use argparser
def set_metadata():
    # Define Total Number of Trials
    totalNumberOfTrials = 10

    #### Sound ####
    # Define Noise Duration
    noiseDuration = 2000 # 2s, Arduino is set in ms
    # Define Punish Tone in Hz
    punishTone = 2000
    # Define Reward Tone in Hz
    rewardTone = 18000

    #### Sucrose ####
    # Define Sucrose Delivery Time
    USDeliveryTime_Sucrose = 5 # 5ms
    # Define Sucrose Consumption Time
    USConsumptionTime_Sucrose = 1000 # 1s, Arduino is set in ms
    #### Airpuff ####
    # Define Airpuff Delivery Time
    USDeliveryTime_Air = 10 # 10ms

    #### Metadata Transfer ####
    # Build metadata list
    metaData = [totalNumberOfTrials, trialNumber, noiseDuration,
    punishTone, rewardTone, USDeliveryTime_Sucrose, USDeliveryTime_Air]
    # Create packet for transfer
    metaData_packet_id = 0
    metaData_packet = txfer.tx_obj(metaData)

    # Return metaData for config and packet for use in transfer
    return metaData, metaData_packet_id, metaData_packet
```
From there I thought I could put each packet into the send function.

Does this look like a way to do it? I'm figuring I'm pretty off base, but I wanted to check in.

PowerBroker2, 2021-04-27:
If you're using a JSON file for configuration management, you can really easily replace a lot of the variable code and clean it up a bit. Also remember that `tx_obj()` doesn't actually send anything, it just stuffs the buffer (probably a bad name choice, but it's too late to change, lmao).

Here's a simplified (and untested) bit of code that should work. I commented it so it should be fairly straightforward to understand, however, if you have any questions, let me know!

```Python
'''

Test JSON file:

{"totalNumberOfTrials": 10,
"trialNumber": 1,
"noiseDuration": 2000,
"punishTone": 2000,
"rewardTone": 18000,
"USDeliveryTime_Sucrose": 5,
"USDeliveryTime_Air": 10,
"USConsumptionTime_Sucrose": 1000}

'''

import JSON
from pySerialTransfer import pySerialTransfer as txfer

if __name__ == '__main__':
    try:
        # initialize the COM port/SerialTransfer connection
        link = txfer.SerialTransfer('COM17')
        
        # read the SON config file
        with open('config.json', 'r') as inFile:
            contents = inFile.read()
            
        # convert from JSON to Dictionary
        config = JSON.loads(contents)
        
        # stuff the TX buffer (https://docs.python.org/3/library/struct.html#format-characters)
        send_len = 0
        send_len += link.tx_obj(config['totalNumberOfTrials'],       send_len, val_type_override='c')
        send_len += link.tx_obj(config['trialNumber'],               send_len, val_type_override='I')
        send_len += link.tx_obj(config['noiseDuration'],             send_len, val_type_override='H')
        send_len += link.tx_obj(config['punishTone'],                send_len, val_type_override='H')
        send_len += link.tx_obj(config['rewardTone'],                send_len, val_type_override='H')
        send_len += link.tx_obj(config['USDeliveryTime_Sucrose'],    send_len, val_type_override='c')
        send_len += link.tx_obj(config['USDeliveryTime_Air'],        send_len, val_type_override='c')
        send_len += link.tx_obj(config['USConsumptionTime_Sucrose'], send_len, val_type_override='H')
    
        # send the data
        link.send(send_len, packet_id=0)
    
    # error handling and garbage collection
    except KeyboardInterrupt:
        try:
            link.close()
        except:
            pass
    
    except:
        import traceback
        traceback.print_exc()
        
        try:
            link.close()
        except:
            pass
```

jmdelahanty, 2021-04-28:
Thanks so much for writing out this example! I'm definitely hoping to use something like JSON to create configuration files so this is perfect.

I understand what it's up to I think, but have run into an error when I execute the program. Here's what it says:
```Python
Traceback (most recent call last):
  File "Documents\gitrepos\headfix_control\python_writing_test.py", line 138, in <module>
    send_len += link.tx_obj(config['totalNumberOfTrials'],       send_len, val_type_override='c')
  File "C:\Users\jdelahanty\.conda\envs\arduino_python\pySerialTransfer\pySerialTransfer.py", line 281, in tx_obj
    val_bytes = struct.pack(self.byte_format + format_str, val)
struct.error: char format requires a bytes object of length 1
```
I tried looking online and found that it could be that it needs to be specifically encoded in a certain format like ```UTF-8``` to pass, but I don't know if that's true or not! Could it be something on the Arduino side? Here's that script just in case:
```C++
// Use package SerialTransfer.h from PowerBroker2 https://github.com/PowerBroker2/SerialTransfer
#include "SerialTransfer.h"

// Rename SerialTransfer to myTransfer
SerialTransfer myTransfer;

const int MAX_NUM_TRIALS = 100; // maximum number of trials possible; much larger than needed but smaller than max value of metadata.totalNumberOfTrials

struct __attribute__((__packed__)) metadata_struct {
  uint8_t totalNumberOfTrials;              // total number of trials for experiment
  uint32_t trialNumber;                     // set trial number for transmission
  uint16_t noiseDuration;                   // length of tone played by speaker
  uint16_t punishTone;                      // airpuff frequency tone in Hz
  uint16_t rewardTone;                      // sucrose frequency tone in Hz
  uint8_t USDeliveryTime_Sucrose;           // amount of time to open sucrose solenoid
  uint8_t USDeliveryTime_Air;               // amount of time to open air solenoid
  uint16_t USConsumptionTime_Sucrose;       // amount of time to wait for sucrose consumption
} metadata;

int32_t trialArray[MAX_NUM_TRIALS]; // create trial array
int32_t ITIArray[MAX_NUM_TRIALS]; // create ITI array

boolean acquireMetaData = true;

void setup()
{
  Serial.begin(9600);
  Serial1.begin(115200);
  
  myTransfer.begin(Serial1, true);
}

void loop(){
  metadata_rx();
}

int metadata_rx() {
  if (acquireMetaData) {
    if (myTransfer.available())
    { 
      myTransfer.rxObj(metadata);
      Serial.println("Received");

      myTransfer.sendDatum(metadata);
      Serial.println("Sent");
      acquireMetaData = false;
    }
  }
}
```

PowerBroker2, 2021-04-28:
Ahhh, right. The answer seems to be here: [https://stackoverflow.com/questions/31551922/python-3-struct-pack-char-format-requires-a-bytes-object-of-length-1](https://stackoverflow.com/questions/31551922/python-3-struct-pack-char-format-requires-a-bytes-object-of-length-1). I may have to do a bug fix and version update for it to work...I'll do it soon

jmdelahanty, 2021-04-28:
That's one of the stack overflow things I found describing the ```UTF-8``` encoding! I still don't really know what's happening in terms of data structure, but encouraging that I happened to stumble into something talking about it haha.

No rush at all! Thanks for looking into that for me.

PowerBroker2, 2021-04-28:
Yeah, I should've used `B` instead of `c`- try with that

jmdelahanty, 2021-04-28:
It made it past that error this time with that change! Now it's giving me a different one from ```tx.Buff``` it seems. Here's what it says:
```Python
Traceback (most recent call last):
  File "C:\Users\jdelahanty\.conda\envs\arduino_python\pySerialTransfer\pySerialTransfer.py", line 437, in send
    self.calc_overhead(message_len)
  File "C:\Users\jdelahanty\.conda\envs\arduino_python\pySerialTransfer\pySerialTransfer.py", line 378, in calc_overhead
    if self.txBuff[i] == START_BYTE:
IndexError: list index out of range
```

PowerBroker2, 2021-04-28:
Ok, looks like I made a couple of silly mistakes:
1. I accidentally capitalized the `json` package
2. I wrongfully assumed the return value of `tx_obj()` was the number of bytes stuffed into the buffer during that call, but it's actually the first value of the next "free" index

Here's the fixed script:

```C++
'''

Test JSON file:

{"totalNumberOfTrials": 10,
"trialNumber": 1,
"noiseDuration": 2000,
"punishTone": 2000,
"rewardTone": 18000,
"USDeliveryTime_Sucrose": 5,
"USDeliveryTime_Air": 10,
"USConsumptionTime_Sucrose": 1000}

'''

import json
from pySerialTransfer import pySerialTransfer as txfer

if __name__ == '__main__':
    try:
        # initialize the COM port/SerialTransfer connection
        link = txfer.SerialTransfer('COM14')
        
        # read the JSON config file
        with open('config.json', 'r') as inFile:
            contents = inFile.read()
            
        # convert from JSON to Dictionary
        config = json.loads(contents)
        
        # stuff the TX buffer (https://docs.python.org/3/library/struct.html#format-characters)
        send_len = 0
        send_len = link.tx_obj(config['totalNumberOfTrials'],       send_len, val_type_override='B')
        send_len = link.tx_obj(config['trialNumber'],               send_len, val_type_override='I')
        send_len = link.tx_obj(config['noiseDuration'],             send_len, val_type_override='H')
        send_len = link.tx_obj(config['punishTone'],                send_len, val_type_override='H')
        send_len = link.tx_obj(config['rewardTone'],                send_len, val_type_override='H')
        send_len = link.tx_obj(config['USDeliveryTime_Sucrose'],    send_len, val_type_override='B')
        send_len = link.tx_obj(config['USDeliveryTime_Air'],        send_len, val_type_override='B')
        send_len = link.tx_obj(config['USConsumptionTime_Sucrose'], send_len, val_type_override='H')
    
        # send the data
        link.send(send_len, packet_id=0)
    
    # error handling and garbage collection
    except KeyboardInterrupt:
        try:
            link.close()
        except:
            pass
    
    except:
        import traceback
        traceback.print_exc()
        
        try:
            link.close()
        except:
            pass
```

jmdelahanty, 2021-04-28:
Hooray! That seems to have sent no problem! One thing that is confusing to me then is how ```send_len``` isn't overwritten with each new link. If we don't use ```+=```, how does it append the values for each variable?

jmdelahanty, 2021-04-28:
Here's something new I'm trying to do. I thought it would be cool to make the packet stuffing happen through a function, but I'm not sure how to do something seemingly basic in Python.  Basically it should read in the ```json``` file do a simple for loop with each entry and ```val_type_override``` and stuff the packet. Here's the function I've tried so far:

```Python
def send_metadata(config, packet_id=0):
    # Read JSON file given to the function and obtain contents
    with open(config, 'r') as inFile:
        contents = inFile.read()
    # Convert from JSON to Dictionary using loads()
    config = json.loads(contents)
    # Give value type override list for sending packets
    # TODO: Add this as part of the configuration file creation and load
    # TODO: Ideally, configuration generation would prohibit user from making
    # config file too big
    val_type_override_list = ["c", "I", "H", "H", "H", "c", "c", "H"]
    # Open transfer link, compile packet, and check packet size
    try:
        # Open comms to Arduino at default baud of 115200 TODO: Grab arduino port automatically
        link.txfer.SerialTransfer('COM12', 115200, debug=True)
        # Stuff TX buffer for packet
        # https://docs.python.org/3/library/struct.html#format-characters
        # Start packet length of 0
        metaData_size = 0
        # For each value in the dictionary and override type
        for (key, value), byte_type in config.items(), val_type_override_list:
            metaData_size = link.txobj(value, metaData_size, val_type_override=byte_type)
        # If the packet is too long, tell experimenter to restructure and exit
        if metaData_size > 244:
            print("Packet is too long to transfer! Restructure config file.")
            print("Exiting")
            sys.exit()
        # Otherwise, send the packet to the Arduino!
        else:
            link.send(metaData_len, packet_id)

        # Transmission is occuring. While the link is unavailable, pass
        while not link.available():
            pass
        # Now, receive the list back from Arduino and confirm its correct
        # Receive metadata from Arduino
        rec_metaData = link.rx_obj(obj_type=type(config),
                                   obj_byte_size=metaData_size,
                                   list_format='i')
        # Check if received metadata is the same as sent metadata
        # If the values in the configuration match received list, move on
        if list(config.values()) == rec_metaData:
            print("Metadata Onboard!")
            # Close link
            link.close()
        # If there's an error, say so and exit program
        # TODO: Introduce error check that will resend several times
        else:
            print("Error! Transfer Unsuccessful...")
            link.close()
            print("Exiting...")
            sys.exit()
```

My for loop is wrong for sure and gives me an error. I've also done this for the for loop, but this obviously gives me the wrong output too:
```Python
for key, value in config.items():
    for byte_type in val_type_override_list:
        metaData_size= link.txobj(value, metaData_size, val_type_override=byte_type)
```
Any idea how I'd do it? It's also obviously fine if it's not a good idea!

PowerBroker2, 2021-04-28:
The one thing I notice is that you are integrating "soft-coded" information (i.e. the real metadata) with "hard-coded" information (`val_type_override_list`). The issue with this is if the user changes either the order of the data in the JSON file or add/subtract/change metadata fields, you would still have to edit the code and might get confusing to maintain.

My suggestion would be to make it all "hard-coded" similarly to how I did in my last code snippet. This is a very, very rare suggestion for me to give since I know "soft-coding" is usually more maintainable. However, this is a special case - making any part of the interface with the config file "soft-coded" will give the person who created the config file enough power to mess up the format of the file/data, which could cause your code to act erratically or crash.

One example of how a user could mess up your program (as in your last reply) would be if they simply changed the order of the data objects in the JSON file. Then, the wrong values and `val_types` would be associated with the wrong experiment parameters.

Another thing I noticed in your latest post is that Python is expecting to receive a list. I'm assuming your Arduino is going to echo the exact data it received from the PC. The rub is that `rx_obj()` can only parse lists of a uniform type and the metadata is not - some are bytes, some are shorts, and some are full ints. I think you'll have to parse out each parameter from the RX buffer individually and compare afterwards. You can do this with a separate dictionary like `rx_metadata`.

Lastly, note that Python dictionaries are not guaranteed to retain or even keep track of the original order of it's elements. For instance, if you have the JSON:

```JSON
('hi': 0, 'bye': 1}
```

There is no guarantee that `dict.values()[0] == 0` or `dict.keys()[0] == 'hi'` will return true. Just something to keep in mind.

PowerBroker2, 2021-04-28:
> My for loop is wrong for sure and gives me an error. I've also done this for the for loop, but this obviously gives me the wrong output too:
> 
> ```C++
> for key, value in config.items():
>     for byte_type in val_type_override_list:
>         metaData_size= link.txobj(value, metaData_size, val_type_override=byte_type)
> ```

Although my suggestion is to not use a for loop, I wanted to explain why this loop doesn't work: If you write out what the loop is doing in words, it makes it a little clearer: "for each value in the config file, loop through each value _type_ and stuff the value in the TX buffer". More simply, the nested loop writes a single `value` `len(val_type_override_list)`-number of times instead of just once.

jmdelahanty, 2021-04-28:
> The one thing I notice is that you are integrating "soft-coded" information (i.e. the real metadata) with "hard-coded" information (val_type_override_list).

Definitely a problem, I had wanted to do this as part of the json file, but wasn't sure how to index that entry in it. I tried to have the ```val_type_override_list``` as one of the entries in the file, but struggled to figure out how to iterate over that list's values while iterating over the metadata entries. 

> The issue with this is if the user changes either the order of the data in the JSON file or add/subtract/change metadata fields, you would still have to edit the code and might get confusing to maintain... One example of how a user could mess up your program (as in your last reply) would be if they simply changed the order of the data objects in the JSON file. Then, the wrong values and ```val_types``` would be associated with the wrong experiment parameters.

This problem is one that I'm naively hoping people wouldn't do... probably not wise to risk it haha

> Another thing I noticed in your latest post is that Python is expecting to receive a list. I'm assuming your Arduino is going to echo the exact data it received from the PC. The rub is that rx_obj() [...]

Okay, got it! Making a ```rx_metadata``` dictionary is what I'll try to do.

> Lastly, note that Python dictionaries are not guaranteed to retain or even keep track of the original order of it's elements.

That is very important to know! I didn't think of that at all!

> Although my suggestion is to not use a for loop, I wanted to explain why this loop doesn't work: If you write out what the loop is doing in words

This is a good idea and definitely helps me understand the loop better. I'll start doing that when writing loops out. Thanks for the tip!

I'll hard-code the data to make sure it can't be tampered with by the user and try to implement the ```rx_metadata``` dictionary today. Thanks PowerBroker!

PowerBroker2, 2021-04-28:
One more tip I have is to make the metadata a sub-element to the JSON file, like this:

```JSON
{{"metadata":   {"config_thing": 123}},
 {"trialArray": [1, 2, 3]},
 {"ITIArray":   [1, 2, 3]}}
```

This will be a minimal impact to the Python code, but will allow for much easier scalability. For instance, now, instead of calling `config['totalNumberOfTrials']`, you'll have to call `config['metadata']['totalNumberOfTrials']`

jmdelahanty, 2021-04-28:
Awesome! I will do this as well!

PowerBroker2, 2021-04-28:
> I tried to have the val_type_override_list as one of the entries in the file, but struggled to figure out how to iterate over that list's values while iterating over the metadata entries.

I personally wouldn't trust the config creator to know what the best `val_type` to use, so I wouldn't include it in the file. However, you could add that info in the JSON like this:

```JSON
{"config_thing": {"value": 123, "val_type": "b"}}
```

Then, you could call:

```Python
metaData_size= link.txobj(config['metadata']['config_thing']['value'], metaData_size, val_type_override=config['metadata']['config_thing']['val_type'])
```

> This problem is one that I'm naively hoping people wouldn't do... probably not wise to risk it haha

If a user can crash your code - no matter how unlikely - they will eventually do the thing that crashes it 😄 

> Thanks PowerBroker!

No problem!

PowerBroker2, 2021-04-28:
Oh, also, check out these tutorials on [JSON](https://www.geeksforgeeks.org/javascript-json/) and [Python dictionaries](https://www.geeksforgeeks.org/python-dictionary/)

jmdelahanty, 2021-04-29:
> I personally wouldn't trust the config creator to know what the best val_type to use, so I wouldn't include it in the file. However, you could add that info in the JSON like this:

If I eventually make a function that would force certain parameters to be the appropriate variable type I'll add it in there. For now, hard-coding it all seems to be a good way to do it.

> If a user can crash your code - no matter how unlikely - they will eventually do the thing that crashes it 😄

Got it haha, I know that I've crashed programs in all kinds of ways that developers probably didn't think I'd do!

> Oh, also, check out these tutorials on JSON and Python dictionaries

Thanks for these resources! 

So after looking through the ```rx_obj``` function a little bit, I thought I would try writing something that receives the data off the Arduino. Here's what I've tried so far:
``` Python
# Receive packet from Arduino
        rxmetaData_size = 0
        rxmetaData_size = link.rx_obj(obj_type='B', start_pos=0, obj_byte_size=rxmetaData_size, byte_format='@')
        rxmetaData_size = link.rx_obj(obj_type='B', start_pos=len(rxmetaData_size), obj_byte_size=rxmetaData_size, byte_format='@')
        # etc...
```
Is this along the lines of a correct implementation? If I understand correctly, the idea is to receive each object similar to how I sent it. I can hard code each variable type for the transmission and then decode the packet after it's all here. In the Arduino file, I'd have to not send the entire ```struct``` back but instead send each piece of it in it's own ```sendDatum``` line. Maybe something like this:
```C++
int metadata_rx() {
  if (acquireMetaData) {
    if (myTransfer.available())
    { 
      myTransfer.rxObj(metadata);
      Serial.println("Received");

      myTransfer.sendDatum(metadata.totalNumberOfTrials);
      // etc...
      Serial.println("Sent Total Number of Trials");
      acquireMetaData = false;
    }
  }
}
```
Is this how you would have someone use it?

PowerBroker2, 2021-04-29:
You're on the right track. Here is something that _should_ work:

```Python
        # Receive packet from Arduino
        rxmetaData = {}
        rxmetaData_size = 0

        rxmetaData['config_thing1`] = link.rx_obj(obj_type='B', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['B']
        rxmetaData['config_thing2`] = link.rx_obj(obj_type='B', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['B']
        # etc...
```

And for the Arduino:

```C++
int metadata_rx() {
  if (acquireMetaData) {
    if (myTransfer.available())
    { 
      myTransfer.rxObj(metadata);
      Serial.println("Received");

      myTransfer.sendDatum(metadata); // both stuffs and sends all data in the correct format in one line

      Serial.println("Sent Metadata");
      acquireMetaData = false;
    }
  }
}
```

jmdelahanty, 2021-04-30:
Alright, I'm trying to finally test it out but I'm running into a problem with my json file that's confusing me. I'm getting this error in my terminal:
```Python
Traceback (most recent call last):
  File "Documents\gitrepos\headfix_control\python_writing_test.py", line 239, in <module>
    config = json.loads(contents)
  File "C:\Users\jdelahanty\.conda\envs\arduino_python\lib\json\__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "C:\Users\jdelahanty\.conda\envs\arduino_python\lib\json\decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "C:\Users\jdelahanty\.conda\envs\arduino_python\lib\json\decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Expecting property name enclosed in double quotes: line 2 column 3 (char 4)
```
I can't see where I don't have the property name enclosed in double quotes! Can you see it?
Here's my json format:
```json
{
  {"metadata": {
  "totalNumberOfTrials": {"value": 10},
  "trialNumber": {"value": 1},
  "noiseDuration": {"value": 2000},
  "punishTone": {"value": 2000},
  "rewardTone": {"value": 18000},
  "USDeliveryTime_Sucrose": {"value": 5},
  "USDeliveryTime_Air": {"value": 10},
  "USConsumptionTime_Sucrose": {"value": 1000}
    }
  }
}
```

jmdelahanty, 2021-05-03:
Good morning Broker of Power!

It looks like this solved the json portion, but I'm now running into a new error in the ```rx_obj``` it seems. Here's what the error is giving me:
```Python
Traceback (most recent call last):
  File "Documents\gitrepos\headfix_control\python_writing_test.py", line 262, in <module>
    rxmetaData['totalNumberOfTrials'] = link.rx_obj(obj_type='B', start_pos=rxmetaData_size)
  File "C:\Users\jdelahanty\.conda\envs\arduino_python\pySerialTransfer\pySerialTransfer.py", line 341, in rx_obj
    buff = bytes(self.rxBuff[start_pos:(start_pos + STRUCT_FORMAT_LENGTHS[obj_type])])
TypeError: 'str' object cannot be interpreted as an integer
```
Am I getting something wrong in the setup with the ```obj_type``` for receiving the data? It appears that the data is being sent to the Arduino no problem.

PowerBroker2, 2021-05-03:
Can you post your entire Python and Arduino programs so I can easily test?

jmdelahanty, 2021-05-03:
Hello! Here you go!

Python:
```Python
# Import JSON for configuration file
import json
# Import pySerialTransfer for serial comms with Arduino
from pySerialTransfer import pySerialTransfer as txfer
# Import sys for exiting program safely
import sys

# Main program

if __name__ == '__main__':
    try:
        # Initialize COM Port for Serial Transfer
        link = txfer.SerialTransfer('COM12', 115200, debug=True)

        #read JSON config file
        with open('C:/Users/jdelahanty/Documents/gitrepos/headfix_control/config.json', 'r') as inFile:
            contents = inFile.read()

        # Convert from JSON to Dictionary
        config = json.loads(contents)

        # stuff TX buffer (https://docs.python.org/3/library/struct.html#format-characters)
        metaData_size = 0
        metaData_size = link.tx_obj(config['metadata']['totalNumberOfTrials']['value'],       metaData_size, val_type_override='B')
        metaData_size = link.tx_obj(config['metadata']['trialNumber']['value'],               metaData_size, val_type_override='I')
        metaData_size = link.tx_obj(config['metadata']['noiseDuration']['value'],             metaData_size, val_type_override='H')
        metaData_size = link.tx_obj(config['metadata']['punishTone']['value'],                metaData_size, val_type_override='H')
        metaData_size = link.tx_obj(config['metadata']['rewardTone']['value'],                metaData_size, val_type_override='H')
        metaData_size = link.tx_obj(config['metadata']['USDeliveryTime_Sucrose']['value'],    metaData_size, val_type_override='B')
        metaData_size = link.tx_obj(config['metadata']['USDeliveryTime_Air']['value'],        metaData_size, val_type_override='B')
        metaData_size = link.tx_obj(config['metadata']['USConsumptionTime_Sucrose']['value'], metaData_size, val_type_override='H')

        link.open()

        link.send(metaData_size, packet_id=0)

        # Receive packet from Arduino
        # Create rxmetaData dictionary
        rxmetaData = {}
        rxmetaData_size = 0

        rxmetaData['totalNumberOfTrials'] = link.rx_obj(obj_type='B', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['B']
        rxmetaData['trialNumber'] = link.rx_obj(obj_type='I', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['I']
        rxmetaData['noiseDuration'] = link.rx_obj(obj_type='H', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['H']
        rxmetaData['punishTone'] = link.rx_obj(obj_type='H', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['H']
        rxmetaData['rewardTone'] = link.rx_obj(obj_type='H', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['H']
        rxmetaData['USDeliveryTime_Sucrose'] = link.rx_obj(obj_type='B', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['B']
        rxmetaData['USDeliveryTime_Air'] = link.rx_obj(obj_type='B', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['B']
        rxmetaData['USConsumptionTime_Sucrose'] = link.rx_obj(obj_type='H', start_pos=rxmetaData_size)
    except KeyboardInterrupt:
        try:
            link.close()
        except:
            pass
    except:
        import traceback
        traceback.print_exc()

        try:
            link.close()
        except:
            pass
```

And here's the Arduino code:
```C++
// Use package SerialTransfer.h from PowerBroker2 https://github.com/PowerBroker2/SerialTransfer
#include "SerialTransfer.h"

// Rename SerialTransfer to myTransfer
SerialTransfer myTransfer;

const int MAX_NUM_TRIALS = 100; // maximum number of trials possible; much larger than needed but smaller than max value of metadata.totalNumberOfTrials

struct __attribute__((__packed__)) metadata_struct {
  uint8_t totalNumberOfTrials;              // total number of trials for experiment
  uint32_t trialNumber;                     // set trial number for transmission
  uint16_t noiseDuration;                   // length of tone played by speaker
  uint16_t punishTone;                      // airpuff frequency tone in Hz
  uint16_t rewardTone;                      // sucrose frequency tone in Hz
  uint8_t USDeliveryTime_Sucrose;           // amount of time to open sucrose solenoid
  uint8_t USDeliveryTime_Air;               // amount of time to open air solenoid
  uint16_t USConsumptionTime_Sucrose;       // amount of time to wait for sucrose consumption
} metadata;

int32_t trialArray[MAX_NUM_TRIALS]; // create trial array
int32_t ITIArray[MAX_NUM_TRIALS]; // create ITI array

boolean acquireMetaData = true;

void setup()
{
  Serial.begin(115200);
  Serial1.begin(115200);

  myTransfer.begin(Serial1, true);
}

void loop(){
  metadata_rx();
}

int metadata_rx() {
  if (acquireMetaData) {
    if (myTransfer.available())
    {
      myTransfer.rxObj(metadata);
      Serial.println("Received Metadata");

      myTransfer.sendDatum(metadata);
      Serial.println("Sent Metadata");
      
      acquireMetaData = false;
    }
  }
}
```

PowerBroker2, 2021-05-03:
The reason your code was erroring was because you weren't parsing any packets with `link.available()`. Also, using relative file paths will help prevent crashes/errors. Here's a script that will work:

```Python
# Import JSON for configuration file
import json
# Import pySerialTransfer for serial comms with Arduino
from pySerialTransfer import pySerialTransfer as txfer
# Import sys for exiting program safely
import sys

# Main program

if __name__ == '__main__':
    try:
        # Initialize COM Port for Serial Transfer
        link = txfer.SerialTransfer('COM12', 115200, debug=True)

        #read JSON config file
        with open('config.json', 'r') as inFile: ############### Relative file path
            contents = inFile.read()

        # Convert from JSON to Dictionary
        config = json.loads(contents)

        # stuff TX buffer (https://docs.python.org/3/library/struct.html#format-characters)
        metaData_size = 0
        metaData_size = link.tx_obj(config['metadata']['totalNumberOfTrials']['value'],       metaData_size, val_type_override='B')
        metaData_size = link.tx_obj(config['metadata']['trialNumber']['value'],               metaData_size, val_type_override='I')
        metaData_size = link.tx_obj(config['metadata']['noiseDuration']['value'],             metaData_size, val_type_override='H')
        metaData_size = link.tx_obj(config['metadata']['punishTone']['value'],                metaData_size, val_type_override='H')
        metaData_size = link.tx_obj(config['metadata']['rewardTone']['value'],                metaData_size, val_type_override='H')
        metaData_size = link.tx_obj(config['metadata']['USDeliveryTime_Sucrose']['value'],    metaData_size, val_type_override='B')
        metaData_size = link.tx_obj(config['metadata']['USDeliveryTime_Air']['value'],        metaData_size, val_type_override='B')
        metaData_size = link.tx_obj(config['metadata']['USConsumptionTime_Sucrose']['value'], metaData_size, val_type_override='H')

        link.open()

        link.send(metaData_size, packet_id=0)
        
        while not link.available():
            pass

        # Receive packet from Arduino
        # Create rxmetaData dictionary
        rxmetaData = {}
        rxmetaData_size = 0

        rxmetaData['totalNumberOfTrials'] = link.rx_obj(obj_type='B', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['B']
        rxmetaData['trialNumber'] = link.rx_obj(obj_type='I', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['I']
        rxmetaData['noiseDuration'] = link.rx_obj(obj_type='H', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['H']
        rxmetaData['punishTone'] = link.rx_obj(obj_type='H', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['H']
        rxmetaData['rewardTone'] = link.rx_obj(obj_type='H', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['H']
        rxmetaData['USDeliveryTime_Sucrose'] = link.rx_obj(obj_type='B', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['B']
        rxmetaData['USDeliveryTime_Air'] = link.rx_obj(obj_type='B', start_pos=rxmetaData_size)
        rxmetaData_size += txfer.ARRAY_FORMAT_LENGTHS['B']
        rxmetaData['USConsumptionTime_Sucrose'] = link.rx_obj(obj_type='H', start_pos=rxmetaData_size)
    
        print(rxmetaData)
    
    except KeyboardInterrupt:
        try:
            link.close()
        except:
            pass
    except:
        import traceback
        traceback.print_exc()

        try:
            link.close()
        except:
            pass
```

jmdelahanty, 2021-05-04:
Oh man, that's embarrassing. I completely forgot about that statement. Thanks! One of our lab members has been imaging a fair bit yesterday and today so later this evening I'll test it to make sure it works. I'll let you know after I add the ```link.available()``` statement.

jmdelahanty, 2021-05-05:
It works! Huzzah! Now onward to send the other arrays! So I've gotten it to send the second array of the three to the Arduino. The Arduino then sends it back. It appears that during this reception, something about the ```while not link.available()``` statement breaks.

Is there something I'm missing? Here's the error:
```Python
Traceback (most recent call last):
  File "Documents\gitrepos\headfix_control\python_writing_test.py", line 257, in <module>
    while not link.available():
  File "C:\Users\jdelahanty\.conda\envs\arduino_python\pySerialTransfer\pySerialTransfer.py", line 534, in available
    self.rxBuff[self.payIndex] = recChar
IndexError: list assignment index out of range
```

And here's what I have immediately after receiving the metadata. ```trialArray``` is generated by a different function that I can include if you'd like:
```Python
# Now send the trial array:
        trialArray_size = 0
        trialArray_size = link.tx_obj(trialArray)
        link.send(trialArray_size, packet_id=1)

        while not link.available():
            pass
```
Note that this is the second ```while not link.available()``` statement. Could that be the issue? The error complains at the first instance of it.

I've also added this into the Arduino code to parse the second packet:
```C++
int trials_rx() {
  if (acquireTrials) {
    if (myTransfer.available())
    {
      myTransfer.rxObj(trialArray);
      Serial.println("Received Trial Array");

      myTransfer.sendDatum(trialArray);
      Serial.println("Sent Trial Array");
      acquireTrials = false;
      acquireITI = true;
    }
  }
}
```
So the loop now looks like this:
```C++
void loop(){
  metadata_rx();
  trials_rx();
}
```

I'm not sure how that would mess with the ```while not link.available()``` step. Any insight?

PowerBroker2, 2021-05-05:
How big is your array in bytes? I'm guessing it's too large

jmdelahanty, 2021-05-05:
I did have the ```MAX_NUM_TRIALS``` array initialized with 100 like so:
```C++
const int MAX_NUM_TRIALS = 100;
```
And then the arrays initialized like this:
```C++
int32_t trialArray[MAX_NUM_TRIALS]; // create trial array
int32_t ITIArray[MAX_NUM_TRIALS]; // create ITI array
```
I literally just changed this to 45, which was I have been previously been able to send as the largest array size from Python and it seems to work now!

PowerBroker2, 2021-05-05:
> I'm looking up now how to change the trialArray to the appropriate size once the metadata has been received.

You could do the following if you need to specify a certain number of bytes to transmit of a single object (if that makes sense):

```C++
myTransfer.sendDatum(trialArray, sendSize);
```

PowerBroker2, 2021-05-05:
> I literally just changed this to 45, which was I have been previously been able to send as the largest array size from Python and it seems to work now!

Ah, yes, cause 32 * 100 > 244 - I forgot you were sending big boi integers

jmdelahanty, 2021-05-05:
Hahaha big boi integers indeed! 

It looks like this all works like it's supposed to! I can now send metadata, trials, and the inter trial intervals all in one command! You're the best PowerBroker! 

I'm going to try and do some of the additional things you suggested like error checking and to try sending information again if something fails to send properly and also automatically collect which port to use for transmitting to the Arduino. I'll also be trying to integrate these functions into a main python function that's going to (hopefully) control everything properly.

Also, side note, I'm using your packages to help another project in the lab! I can email you about it if you'd like so you can add that to your work too. Basically, a computer vision algorithm will identify which behavior an animal is doing and, from that, send a trigger to an Arduino that will (for now) turn on a color coded LED. If you want me to send you more in depth stuff let me know!

PowerBroker2, 2021-05-05:
> It looks like this all works like it's supposed to! I can now send metadata, trials, and the inter trial intervals all in one command! You're the best PowerBroker!

Awesome, glad to help!

> Also, side note, I'm using your packages to help another project in the lab! I can email you about it if you'd like so you can add that to your work too. 

Yeah man! I'm always interested in how my code is used. Learning about stuff like this is super rewarding as an open-source programmer

jmdelahanty, 2021-05-05:
I'll write you an email now! I'll close the issue as well since things are working.

closed 2021-05-05 by jmdelahanty