PowerBroker2/pySerialTransfer #35: How to transmit image from esp32-cam to python

arelhabib, 2021-07-08:
Hello PowerBroker2!
Recently i want to made a project weight scale with esp32-cam taking the picture. the weight scale script is done by simple serial print and gui serial monitor with python. but had problem with transmitting the image because esp32cam (ai thinker) im using only have spare gpio pinout with adc channel not work when using the wifi board [mentioned in here](https://github.com/espressif/arduino-esp32/issues/102#issuecomment-295272351)
Then i found this [post](https://forum.arduino.cc/t/transfer-pictures-from-esp32-cam-to-esp32-via-serial/647488/59), so in tx side i made `if else` separating the scale and image transmit, with image transmit function copied from the post. then in rx side i make signal to stop reading the scale and send a `char` to esp32cam to start transmit the images
But based the post, how i receive that splitted data in python? did it also need `struct `to receive or there's another way to do that?

PowerBroker2, 2021-07-08:
While this is possible to do with SerialTransfer/pySerialTransfer, I think it would be easier to use FTP to transfer the pictures over WiFi to the computer running Python. Have you tried this?

The API between the Arduino and Python libraries are very similar and the RX sketch in the post you referenced should largely work for the Python script (minus some updates). Give it a shot and if you have issues, feel free to post your Arduino and Python code along with your issues. You might get better/more instantaneous help by posting on the Arduino forum. Library GitHub issues are more for bug reports and enhancement suggestions.

arelhabib, 2021-07-10:
> While this is possible to do with SerialTransfer/pySerialTransfer, I think it would be easier to use FTP to transfer the pictures over WiFi to the computer running Python. Have you tried this?

Oh i forgot to say that i also use i2c lcd to display the scale result, and when i try using ftp, the lcd not displaying anything (sorry for dark photo)
[image]
and it always work when wifi is unused, i dont know its because not much power or because the gpio problem i mentioned above

> The API between the Arduino and Python libraries are very similar and the RX sketch in the post you referenced should largely work for the Python script (minus some updates). Give it a shot and if you have issues, feel free to post your Arduino and Python code along with your issues. 

From the arduino post, im changing the tx code  myTransfer using serial to read directly to pc
```c
  Serial.begin(115200);                     // Define and start serial monitor
  // 115200,256000,512000,962100
  //Comm.begin(962100, SERIAL_8N1,15,14);     //, Comm_Txd_pin, Comm_Rxd_pin); // Define and start Comm serial port
  myTransfer.begin(Serial);
```
and here the loop
```C
void loop() {
  if (Serial.available() > 0)
  {
    char incomingByte = Serial.read();
    if (incomingByte == '1')
    {
      cam();
    }
  }
  else
  {
  }
  char incomingByte = '0';
  weight();
}
```
and i think its worked well (that `0,0` below is the weight scale result)
[image]

and in the python rx side im going to make signal using command from button gui to stop weight scale monitor, send that character to espcam and start receving image bytes, but im not write that yet because im still dont know how to restructure the image transferred. 
btw, is it possible to just take the picture and send via regular serial without using this library then python just receive that image, or with/without this library its need to split the image, transmit it bit by a bit and the end is python need to restructure the image?

Actually im borrowing esp board from my friend to try arduino code from the post with SerialTransfer 1.4.1 and change a bit in rx side to make it work.
the result is sometimes it getting payload error in the first transmit or crc error in the middle transmit resulting image corrupted
[image]
is it possible because the jumper cable not good or power that needed is not enough or something else? 
image in the left side is using phone adapter and the right side is using usb cable, but the right side is much of replace because the same name. im using ftp server in my android because already have the app 
>You might get better/more instantaneous help by posting on the Arduino forum. Library GitHub issues are more for bug reports and enhancement suggestions.

Ahh, sorry about that, i just dont know where to ask actually, and i think arduino forum is only talk about arduino thing. Maybe i need to make an account and visit the forum, but thankyou for answering my questions:grin:

PowerBroker2, 2021-07-17:
> Actually im borrowing esp board from my friend to try arduino code from the post with SerialTransfer 1.4.1 and change a bit in rx side to make it work.

I would suggest using lib version 2.1.2 and basing your code off of the examples [uart_tx_file.ino](https://github.com/PowerBroker2/SerialTransfer/blob/master/examples/uart_tx_file/uart_tx_file.ino) and [uart_rx_file.ino](https://github.com/PowerBroker2/SerialTransfer/blob/master/examples/uart_rx_file/uart_rx_file.ino) instead of what is on the old forum post.

> im still dont know how to restructure the image transferred.

The examples I linked above restructures a file and you should be able to learn how to from reading the comments.

arelhabib, 2021-07-24:
sorry for late reply, btw my ftp transfer attempt did not working was because of not much power to the lcd and i forgot to initialize the lcd itself in the arduino code. sorry about pinging you in the github issues for my silliness.
maybe im gonna ask you again if im using this method in the future, thank you for helping me😁

closed 2021-07-24 by arelhabib