PowerBroker2/pySerialTransfer #37: Equivalent examples in python for the given examples in the Arduino Library

AlexanderPeters, 2021-07-21:
It would be nice to see an equivalent examples folder in this repo that matches the functionality of the great examples given for the Arduino SerialTransfer library. The Arduino examples are great, but sometimes its a struggle to get the corollary code working in python.

PowerBroker2, 2021-07-22:
Because of how Python works there definitely are slight nuances between the two libs, but browsing through the source files should easily answer most questions (I try to write thorough doc strings). However, I will seriously consider making more Python example scripts

renderedSafe, 2021-08-14:
I second this. It's unclear to me how the struct parsing on the arduino side happens and what to send on the Python side to get parsed into a struct.

PowerBroker2, 2021-08-14:
I had some extra time today, so I whipped up some examples. @renderedSafe, do these examples answer your questions? If so, I'll tag out a new version.

renderedSafe, 2021-08-15:
@PowerBroker2 those examples are great, thank you. Not just for that, but for the whole solution you've written. I'm working on some robotics stuff and really needed to pass information reliably between a Raspberry Pi and an Arduino mega, so I was happy when I stumbled across your project. 

I've been on a hiatus from Python for a bit while working with other languages, so I'm not quite readjusted, but there seems to be a lot of boilerplate to stuff the packet with our data that unfortunately seems to stem from Python's weak/dynamic typing. I'll be using this package in my project, and if I come up with a fix you may see a PR from me with some utility functions that abstract some of the boilerplate away if possible.  

Thanks again for your active and open development.

PowerBroker2, 2021-08-15:
No problem! Always glad to help and appreciate all feedback. Nothing is more satisfying than getting real engagement with the open source community. I'll be sure to look at your PR if/when you submit.

closed 2021-08-15 by PowerBroker2