PowerBroker2/pySerialTransfer #38: Possible off by one bug.

AlexanderPeters, 2021-07-21:
I get the following error if data was previously sent with the maximum allowed size.

e.x. (Sending data from Arduino)
myTransfer.sendData(MAX_PACKET_SIZE);

Results in: (Receiving data on Raspberry Pi)
Traceback (most recent call last):
  File "serialReadTest2.py", line 16, in <module>
    while not link.available():
  File "/home/pi/.local/lib/python3.7/site-packages/pySerialTransfer/pySerialTransfer.py", line 537, in available
    self.rxBuff[self.payIndex] = recChar
IndexError: list assignment index out of range

From the following code:
link = pySerialTransfer.SerialTransfer('serial0')
link.open()

while not link.available():

This issue does not occur if data is sent with a size one less than that of the MAX_PACKET_SIZE.

I'm not sure if this is a bug, or if I'm stupid.
Thanks

PowerBroker2, 2021-07-22:
Can you post a full example python script that exemplifies your issue?

AlexanderPeters, 2021-07-22:
See attached for send and receive scripts, let me know if you have any questions.
The following is the complete output of when the python script is run after the Arduino has turned on and begun sending data.

Starting
Traceback (most recent call last):
  File "serialReadTest2.py", line 16, in <module>
    while not link.available():
  File "/home/pi/.local/lib/python3.7/site-packages/pySerialTransfer/pySerialTransfer.py", line 537, in available
    self.rxBuff[self.payIndex] = recChar
IndexError: list assignment index out of range

[Example.zip](https://github.com/PowerBroker2/pySerialTransfer/files/6859479/Example.zip)

PowerBroker2, 2021-07-22:
This is most likely caused by having a packet payload length byte value of 255. My guess is the Arduino code is trying to stuff too many bytes into the packet payload than it can handle. That coupled with the outdated API you're using (at least on the Arduino side) can cause problems.

My suggestion is to look at [this issue](https://github.com/PowerBroker2/SerialTransfer/issues/72) for more info on how to transfer a file and forget about the original Arduino forum post. Unfortunately the post was made read only...

PowerBroker2, 2021-08-29:
I'm fairly certain this was fixed in an earlier version, going to close this now, but let me know if the issue still persists

closed 2021-08-29 by PowerBroker2