PowerBroker2/pySerialTransfer #40: Getting crc errors as soon as I am not looping if myTransfer.available()

joonasstenvall, 2021-07-31:
Hi, I'm trying to send a list of xbox controller values. I'm sure there could be a better option but I am not smart enough to figure that out. The list is currently 21 indexes long and I can send that to the arduino no problem.
But as soon as I try to do something outside of receiving serialtransfer packets, the code breaks. Could you help me a bit?

The python code.
```
from evdev import list_devices, InputDevice, categorize, ecodes, KeyEvent,event_factory
import time
from pySerialTransfer import pySerialTransfer as txfer

gamepad = InputDevice('/dev/input/event0')
analogValue = [0,0,0,0,0,0]
buttonValue = [0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]
stickIndexes = [0,1,3,4]
buttonIndex = [304,305,307,308,310,311,314,315,316,317,318]
currentstate = [0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]
oldstate = [0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]
send_size = 0
list_size = 0

def map2(input,inmin,inmax,outmin,outmax):
	return int((input-inmin)*(outmax-outmin)/(inmax-inmin)+outmin)

def checkKeys():
	for x in gamepad.active_keys():
				for y in range(len(buttonIndex)):
					if x == buttonIndex[y]:
						buttonValue[y] = 1
	#print("checked keys")

def checkAnalog():
	for x in range(4):
		if gamepad.absinfo(stickIndexes[x]).value > 2000 or gamepad.absinfo(stickIndexes[x]).value < -2000:
			analogValue[x] = map2(gamepad.absinfo(stickIndexes[x]).value,-32767,32767,-1000,1000)
		else:
			analogValue[x] = 0

		analogValue[4] = gamepad.absinfo(2).value
		analogValue[5] = gamepad.absinfo(5).value

	if gamepad.absinfo(16).value != 0:
		if gamepad.absinfo(16).value == -1:
			buttonValue[11] = 1
		elif gamepad.absinfo(16).value == 1:
			buttonValue[12] = 1

	if gamepad.absinfo(17).value != 0:
		if gamepad.absinfo(17).value == -1:
			buttonValue[13] = 1
		elif gamepad.absinfo(17).value == 1:
			buttonValue[14] = 1
	#print("checked analog")

if __name__ == '__main__':
	try:
		link = txfer.SerialTransfer('/dev/serial0')
		
		link.open()
		time.sleep(2) # allow some time for the Arduino to completely reset
		
		while True:
			send_size = 0
			
			###################################################################
			# Send a list
			###################################################################

			checkKeys()
			checkAnalog()
			print(analogValue+buttonValue)
			

			list_ = analogValue + buttonValue
			list_size = link.tx_obj(list_)
			send_size += list_size
			buttonValue = [0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]
			
			###################################################################
			# Transmit all the data to send in a single packet
			###################################################################
			link.send(send_size)
			
			###################################################################
			# Wait for a response and report any errors while receiving packets
			###################################################################
			
			while not link.available():
			#if False:
				if link.status < 0:
					if link.status == txfer.CRC_ERROR:
						print('ERROR: CRC_ERROR')
					elif link.status == txfer.PAYLOAD_ERROR:
						print('ERROR: PAYLOAD_ERROR')
					elif link.status == txfer.STOP_BYTE_ERROR:
						print('ERROR: STOP_BYTE_ERROR')
					else:
						print('ERROR: {}'.format(link.status))
			
			###################################################################
			# Parse response list
			###################################################################
			#rec_list_  = link.rx_obj(obj_type=type(list_),
										#obj_byte_size=list_size,
										#list_format='i')
			###################################################################
			# Display the received data
			###################################################################
			#print('SENT: {}'.format(list_))
			#print('RCVD: {}'.format(rec_list_))
			#print(' ')
			link.close()
    
	except KeyboardInterrupt:
		try:
			link.close()
		except:
			pass

	except:
		import traceback
		traceback.print_exc()
		
		try:
			link.close()
		except:
			pass
```
Arduino code

Incoming list variables

```
const int xboxInputAmount = 21; //How many xbox values there will be
int32_t inputArray[xboxInputAmount]; //This will contain every xbox controller outputs

const int xbLSTICKAX = 0;//Left stick X axis
const int xbLSTICKAY = 1;//Left stick Y axis
const int xbLBA = 4;//Left trigger axis
const int xbRSTICKAX = 2;//Right stick X axis
const int xbRSTICKAY = 3; //Right stick Y axis
const int xbRBA = 5;//Right trigger axis
const int xbA = 6;//Button A
const int xbB = 7;//Button B
const int xbX = 8;//Button X
const int xbY = 9;//Button Y
const int xbLB = 10;//Button Left Bumber
const int xbRB = 11;//Button Right Bumber
const int xbSHARE = 12;//Button Share
const int xbSETTINGS = 13;//Button Settings
const int xbEMBLEM = 14;//Button Emblem
const int xbLSTICKD = 15;//Left stick down
const int xbRSTICKD = 16;//Right stick down
const int xbDL = 17;//Dpad left
const int xbDR = 18;//Dpad Right
const int xbDU = 19;//Dpad up
const int xbDD = 20;//Dpad down
```

```
void setup()
{
	Serial1.begin(115200);
	Serial.begin(115200);
	Serial.println("Start");
	myTransfer.begin(Serial1);

}
```

Main and the  code I am having troubles with.
```
void buttonsCheckMenuNavigation()//Menu navigation code
{
	if (inputArray[xbA])
	{
		menu_system.call_function(select);//Calls function attached to select
	}

	if (inputArray[xbDL])
	{
		menu_system.call_function(decrease);//Calls function to decrease current value
	}

	if (inputArray[xbDR])
	{
		menu_system.call_function(increase);//Calls function to increase current value
	}

	if (inputArray[xbDU])
	{
		menu_system.switch_focus(false);//Scrolls menu up
	}
	if (inputArray[xbDD])
	{
		menu_system.switch_focus(true);//Scrolls menu down
	}
}
void checkTransfer()
{
	if (myTransfer.available())
	{
		Serial.println("Check transfer");
		uint16_t recSize = 0;
		recSize = myTransfer.rxObj(inputArray, recSize); 

		//Increase and decrease speed value
		if (inputArray[xbLBA] < inputArray[xbRBA])
		{
			//Right trigger code
			speedmodifier = map(inputArray[xbRBA], 0, 1000, 0, 100);
			speed = speed + speedmodifier;
			//Checks if speed is over max
			if(speed > MaxTotalSpeed){
				speed = MaxTotalSpeed;
			}
		}
		else
		{
			//Left trigger code
			speedmodifier = map(inputArray[xbLBA], 0, 1000, 0, 100);
			speed = speed - speedmodifier;
			//If speed is negative
			if(speed < 0){
				speed = 0;
			}
		}
		//Serial.println(speed);

		//Maps stick analog values to the speed
		for (int i = 0; i < 4; i++)
		{
			inputArray[i] = map(inputArray[i], 0, 1000, 0, speed);
		}

		//Print inputArray
		
		for(int i = 0; i < xboxInputAmount; i++){
			Serial.print(inputArray[i]);
			Serial.print("|");
		}
		Serial.println();

		for(uint16_t i=0; i < myTransfer.bytesRead; i++)
			myTransfer.packet.txBuff[i] = myTransfer.packet.rxBuff[i];

		myTransfer.sendData(myTransfer.bytesRead);
		
	}
}

void loop()
{
	checkTransfer();
	buttonsCheckMenuNavigation();

}
```

Errors are
ERROR: STOP_BYTE_ERROR
ERROR: CRC_ERROR
ERROR: STALE PACKET

Also the menu code should work, I'm having troubles with sending values and acquiring the xbox button presses but I think I can fix that.

PowerBroker2, 2021-08-01:
If you're having trouble using the library, the best way to get help is by asking on the Arduino forum or Stack Exchange. GitHub issues are more for notifying users of bugs or to suggest enhancements.

That being said, one tip I would give is to not call `link.close()` every loop.

closed 2021-08-01 by PowerBroker2

joonasstenvall, 2021-08-01:
> 
> 
> If you're having trouble using the library, the best way to get help is by asking on the Arduino forum or Stack Exchange. GitHub issues are more for notifying users of bugs or to suggest enhancements.
> 
> That being said, one tip I would give is to not call `link.close()` every loop.

My bad, I was thinking that the best place to get an answer would be from the guy who wrote it. :)

PowerBroker2, 2021-08-01:
Nope, I'm too busy with all my other projects to deep dive into other projects. Those on the forums voluntarily do that and do it very well.