PowerBroker2/pySerialTransfer #44: pySerialTransfer Questions

hawkseng, 2021-11-28:
Hello PowerBroker2,

We are a high school robotics team that came across your code.
One of our goals is to run serial communication between Python on a laptop and an Arduino over RS485.
We have tried your pySerialTransfer code to verify functionality.

We are able to get Python to send and receive data while using the USB connection to the Arduino, but not over RS485.  At first, we thought our RS485 communication could be a hardware issue.
However, all of our hardware appears to work individually.

1.  We have tested our USB to RS485 adapter using a second adapter and two instances of "AccessPort" from Serialcomm.com.  We are able to communicate between two laptops with this method.

2.  We replaced the USB to RS485 adapters using two Arduinos connected through their serial pins.  The Arduinos both run an "empty" sketch with pins 0 and 1 set as inputs.  We are able to communicate between two laptops with AccessPort as above.

3.  We have tested two Arduinos communicating over RS485 using TTL to RS485 converters.  These work as well.

4.  We have tried using a USB to RS485 adapter with A, B and ground running to the TTL to RS485 converter.  We are using Serial1 for RS485 communication and Serial to send debug statements to the Serial Monitor.  The code never enters the "if(myTransfer.available())" loop on the Arduino.

5.  To try to simplify the connections, our latest test was to eliminate the RS485 connections.  We returned to using two Arduinos.  One Arduino used the setup described in "2" above to serve as a USB to TTL converter.  The second Arduino ran your sample code.  The result is the same as in "4" above.  In both of these cases, we modified the code to use Serial1.  After the "if(myTransfer.available())" we added "else Serial.println("Not available");" to give us feedback.

After reading through several web pages that you have offered advice on, we came across issue #32 (https://github.com/PowerBroker2/pySerialTransfer/issues/32).
We tried several of the code examples you provide there with no luck.

Software is our weakest area this year.  The addition of Python has made it more complicated for us.
We apologize if there is a simple solution that we are missing.

We are not understanding much of the code as it relates to SerialTransfer and pySerialTransfer, so we do not know how to further analyze the "if(myTransfer.available())" functionality.  We have started reading through "Serial Input Basics" and your "Serial Input Advanced" tutorials on the Arduino website.  

Another issue that came up for us was in trying one of your examples shared here: (https://forum.arduino.cc/t/arduino-to-arduino-serial/621141/15).  Our Arduino would not compile your code and gave us the following error: "class SerialTransfer' has no member named 'txBuff'"

With all that said, we are wondering if you have any suggestions since pySerialTransfer seems like it would be helpful to us.

Thanks in advance for your time.

hawkseng, 2021-11-28:
To add a bit of clarity, we did use an Arduino Uno as our USB to TTL converter.
An Arduino Mega is our device receiving data from Python.

In "4" above, we were able to verify that "something" was happening on the Arduino Rx pin using an oscilloscope.
We have not yet tried using a logic analyzer on this pin to verify the data.
However, with all of our tests combined, we are hoping that we are missing something that might be obvious to others.

Thanks again.

PowerBroker2, 2021-11-28:
> We are able to get Python to send and receive data while using the USB connection to the Arduino

Is that in general, or with the libraries?

> two instances of "AccessPort" from Serialcomm.com

I'm not familiar with Serialcomm's products, so I'm not sure how to help interface with them.

> We have tested two Arduinos communicating over RS485 using TTL to RS485 converters. These work as well.

Is that with the libraries?

> We have started reading through "Serial Input Basics" and your "Serial Input Advanced" tutorials on the Arduino website.
>
>  Another issue that came up for us was in trying one of your examples [...]

Yeah, the library has changed a bit since I wrote "Serial Input Advanced" and that one Arduino forum post, so the error is to be expected.

### Best way to get things to work:

1. Verify the bauds match (if using SoftwareSerial.h, don't use a baud higher than 9600)
2. Verify the correct serial ports are being used
3. Verify the RS485 converters are either full duplex, or are configured correctly (TX only or RX only)
2. Next, connect one Arduino to one computer via USB. Run [this example Python script](https://github.com/PowerBroker2/pySerialTransfer/blob/master/examples/data/Python/rx_data.py) (with the correct COM port) and [this example Arduino sketch](https://github.com/PowerBroker2/pySerialTransfer/blob/master/examples/data/Arduino/tx_data/tx_data.ino).
2. Once you've verified the previous step works, introduce the RS485 converters and test the same code (update the serial ports on the Arduino if needed)
3. Once you get the previous step to work, you should be good to go!

If you can't get things working based on the steps above, I'd suggest posting the code here AND a neatly drawn schematic.

> To add a bit of clarity, we did use an Arduino Uno as our USB to TTL converter.

That might be an issue. I highly suggest spending $10 or so and get a decent USB to TTL converter on Amazon. In fact, you might just get a USB to RS485 converter similar to [this](https://www.amazon.com/Industrial-USB-RS485-Converter-Communication/dp/B081MB6PN2/ref=sr_1_3?keywords=usb+to+rs485&qid=1638131365&qsid=142-7961220-2896356&sr=8-3&sres=B081MB6PN2%2CB076WVFXN8%2CB0195ZD3P4%2CB07L2VLY5D%2CB078X5H8H7%2CB005CPI0JQ%2CB010KJSCR8%2CB07SD65BVF%2CB00NKAJGZM%2CB08SM5MX8K%2CB001Q7X0XK%2CB08T5YVP7V%2CB07B416CPK%2CB08PL23NW8%2CB081NBCJRS%2CB083169369&srpt=ELECTRONIC_ADAPTER) (be aware that this converter is half duplex).

### How `available()` works (sort of):

This is oversimplified, but `available()` reads all bytes from the specified UART port's input buffer and attempts to parse the received bytes into a coherent packet. If the parsing is successful and a full, intact packet is received, `available()` will return true and you can do whatever you want with the newly arrived data. If only part of a packet has arrived, it will silently return false and wait for more bytes. If an error arises (i.e. CRC calculation mismatch, unexpected number of payload bytes, etc), it will return false with an error. That error can be accessed by checking the `status` member.

Note: Call `available()` as frequently as possible and do not use blocking code whatsoever.

hawkseng, 2021-11-28:
Thank you for the quick response.
Regarding your comments:
 - Our Python to Arduino communication over USB was with your pySerialTransfer libraries installed.

- The test of our USB to RS485 adapters was done using AccessPort.  This just gave a terminal connected to the COM ports of our two adapters.  This allowed us to send simple text between the laptops, telling us that the USB to RS485 adapters are functioning.  We are working with half duplex adapters.

- Our two Arduinos over RS485 was not with your libraries.  We used code found online for this purpose.  This tells us that our TTL to RS485 adapters are working properly.

- Thanks.  It is a relief that the "txBuff" issue is not on our end.

- Our baud rates do match and we have stayed at 115200 for everything, but will verify again.

- We are certain our COM ports are correct, but will verify these as well.

- The RS485 adapters are half duplex.  The USB to RS485 is taken care of by USB.  We did have the RS485 to TTL configured as RX for our initial tests, using your "datum" code.

- We will get back to you on suggestions 4 and 5.

- We do not intend to use USB to TTL conversion, so were not planning to purchase that adapter.  We do know that we can send data between two Arduinos this way, based on our second test with AccessPort.  Since these worked for simple text between two laptops, we reasoned that the Arduinos should be able to communicate this way if the Mega is switched to Serial1 and runs your code.  The Uno simply converts USB to TTL and sends to the Serial1 pins.  At least that is our hope.

We will get back to you after we try your suggestions, and look forward to the "good to go" portion of suggestion 6.

We have been running our Python code through Spyder.  Since this worked using the USB connection, we reasoned that it should also work over RS485.  Perhaps this is not the case.  We will have to test that as well.

Thank you again for your time.

hawkseng, 2021-11-30:
Hello Again PowerBroker2,

Here is what we learned today:
Suggestions 4 and 5 worked perfectly.
However, "good to go!" was not quite there since we still were not able to send data to the Arduino.
As it turns out, the RS485 to TTL converter was the issue - it could transmit but not receive, likely due to us.
(How easy it would have been to swap that out earlier!)
With that, we were able to send data and receive data using your example code.

Our next step (later in the week) is to modify the Arduino code to control the driver and receiver enable pins on the RS485 module.  We tried a quick modification to your Arduino sketch (https://github.com/PowerBroker2/pySerialTransfer#example-arduino-sketch) by enabling the module as a receiver in the setup.  We also enable the module as a receiver at the beginning of the "if" statement in the void loop.  We enable the module as a transmitter before the "myTransfer.sendData" command.

With this modification, we are able to send data from Python to the Arduino and back to Python.  However, we are only able to do this once.  We added a print statement to the "while True:" loop in the Python code (https://github.com/PowerBroker2/pySerialTransfer#example-python-script).  Python does begin that loop a second time.

Our next step is to look into timing our enable line.  If we can assume all data has been received by the Arduino after the "for" loop and all data has been sent after "myTransfer.sendData," is it reasonable that Python is attempting to send data on the next loop before our enable line has changed?  

Besides our issue with the RS485 to TTL converter, we also realized that running our Uno as a USB to TTL converter and communicating over the serial lines to the Mega did not work in either direction.  Since this is not our goal, it is not a priority to find out why this does not work.  However, we did want to note that we did try it again.

Thank you again for your feedback, it helped us to find the problem that should have been easy to find in our troubleshooting!

PowerBroker2, 2021-12-01:
Glad you got it to work!

Out of curiosity, which high school robotics team are you associated with?

hawkseng, 2021-12-01:
Yes, unfortunately we should have figured that issue out on our own!
Thank you for the external set of "eyes" on the issue.

We are at Hoffman Estates High School outside of Chicago.
We are working on an ROV for the MATE ROV competition:
https://materovcompetition.org/
Our plan is to use RS485 through our tether - something our school has not tried yet.

closed 2021-12-01 by PowerBroker2