PowerBroker2/pySerialTransfer #46: Problems with pySerialTransfer

BastianDohle, 2021-12-14:
Hi PowerBroker,

i want to transfer data from python to an arduino nano clone. Ive used your example sketches and changed them just a little bit. Im able to transfer one value but then it stops sending. Im using python 3.9 and SerialTransfer version 2.1.6 i think. 

Python:
`from time import sleep
from pySerialTransfer import pySerialTransfer as txfer
import time

class struct(object):   #data i want to transfer
    gear_f = 16383
    gear_r = 0
    gear_l = 0

if __name__ == '__main__':
    try:
        flight_data = struct
        link = txfer.SerialTransfer('COM8')

        link.open()
        sleep(5)

        while True:
            sendSize = 0

            sendSize = link.tx_obj(flight_data.gear_f, start_pos=sendSize)   #only sends once
            sendSize = link.tx_obj(flight_data.gear_r, start_pos=sendSize)   #doesnt send at all
            sendSize = link.tx_obj(flight_data.gear_l, start_pos=sendSize)   #doesnt send at all

            print("sending")
            link.send(sendSize)

            while not link.available(): #goes in this loop but not in if statement
                if link.status < 0:
                    print('ERROR: {}'.format(link.status))

    except KeyboardInterrupt:
        link.close()    #compiler message: Name 'link' can be undefined`

Arduino:
#include "SerialTransfer.h"

SerialTransfer myTransfer;

struct STRUCT{
   int gear_f;
   int gear_r;
   int gear_l;
} flight_data;

//int x=0;
//int y=0;
//int z=0;
const int ledpin=9;

void setup()
{
  Serial.begin(115200);
  myTransfer.begin(Serial);
  pinMode(ledpin, OUTPUT);
}

void loop()
{
  if(myTransfer.available())
  {
    // use this variable to keep track of how many
    // bytes we've processed from the receive buffer
    uint16_t recSize = 0;

    recSize = myTransfer.rxObj(flight_data, recSize);
    
    //recSize = myTransfer.rxObj(x, recSize);
    //recSize = myTransfer.rxObj(y, recSize);
    //recSize = myTransfer.rxObj(z, recSize);

    if(flight_data.gear_f==16383)
      {
        digitalWrite(ledpin, HIGH);  
      }
  }
}

PowerBroker2, 2021-12-15:
Try using `int32_t` in the Arduino code - Python and Arduino do not usually have the same int lengths

PowerBroker2, 2021-12-17:
No response from OP, guessing problem was resolved

closed 2021-12-17 by PowerBroker2

BastianDohle, 2021-12-20:
Hi, Ive tested it and it works fine now. Thank you.  Von: PB2Gesendet: Friday, December 17, 2021 8:01 PMAn: PowerBroker2/pySerialTransferCc: BastianDohle; AuthorBetreff: Re: [PowerBroker2/pySerialTransfer] Problems with pySerialTransfer (Issue #46) No response from OP, guessing problem was resolved