PowerBroker2/pySerialTransfer #47: CRC ERROR keep occuring

jaykorea, 2022-01-04:
Hello powrebroker

First of all really big thanks to your awesome library.

As I'm using your library for transmitting simulator data (python) to arduino(mega) CRC Error keep occur and pySerialTransfer Thread keep waiting for timeout and send CRC Error.
I lower the baud rate to 9600 and provide some delay to work properly

here Is my code 
[image]

jaykorea, 2022-01-04:
[image]
[image]

and also this is My Python codes

PowerBroker2, 2022-01-04:
For debugging purposes, create a simple Python and Arduino programs that replicate the data transfer in your original project. Having simpler programs will make debugging a lot easier.

If you are still having trouble finding the error, feel free to post the simplified programs here (NOT pictures - inline code text using markdown tags) along with a detailed description of what exactly is going wrong.

jaycho112, 2022-01-07:
sorry for my ignorance..

After 5 minutes  I send float data to arduino, CRC Error keep occuring.

I think that USB TTL module works wrong, so I Changed it to arduino mega serial0 port to transmit directly without TTL module

However unfortunately, Error keep occuring

Here is my python code, and I run thread to send data

   def transmit_data(self):
        try:
            sendsize = 0

            sendsize = self.link.tx_obj(self.vehicle_speed, start_pos=sendsize)
            sendsize = self.link.tx_obj(self.vehicle_throttle, start_pos=sendsize)
            sendsize = self.link.tx_obj(self.vehicle_rpm, start_pos=sendsize)
            sendsize = self.link.tx_obj(self.vehicle_accelX, start_pos=sendsize)
            sendsize = self.link.tx_obj(self.vehicle_accelY, start_pos=sendsize)
            sendsize = self.link.tx_obj(self.vehicle_lat, start_pos=sendsize)
            sendsize = self.link.tx_obj(self.vehicle_lon, start_pos=sendsize)

            self.link.send(sendsize)

        except KeyboardInterrupt:
            self.link.close()

 def serialThread():
    # global world_flag
    global world
    global p
    global v
    global c
    st = SerialTransmitter()
    st.connectToMCU() #Connect to arduino port
    st.link.open()
    time.sleep(5) # allow some time for the Arduino to completely reset
    while True:
        st.store_data(world,p,v,c)
        st.printdata()
        st.transmit_data()
        time.sleep(0.1)

And this is my arduino code

    if(myTransfer.available())
  {
    // use this variable to keep track of how many
    // bytes we've processed from the receive buffer
    uint32_t recSize = 0;
//    Serial_arr[0] = {'/0'};

    recSize = myTransfer.rxObj(ds, recSize);
    Serial3.print(ds.temp_vehicle_speed);
    Serial3.print(" ");
    Serial3.print(ds.temp_vehicle_throttle);
    Serial3.print(" ");
    Serial3.print(ds.temp_vehicle_rpm);
    Serial3.print(" ");
    Serial3.print(ds.temp_vehicle_accelX);
    Serial3.print(" ");
    Serial3.print(ds.temp_vehicle_accelY);
    Serial3.print(" ");
    Serial3.print(ds.temp_vehicle_lat);
    Serial3.print(" ");
    Serial3.print(ds.temp_vehicle_lon);
    Serial3.print(" ");    
    Serial3.print(" | ");
    Serial3.println(" ");
}

I've used the code very simply , but I think thread interrupt some function of your library. 
plz let me know what's the problem 

Thank you powerbroker

PowerBroker2, 2022-01-07:
For debugging simplify your code as much as possible before complicating it (i.e. don't use threading - just transfer data). In fact, I would suggest using [some built-in examples](https://github.com/PowerBroker2/pySerialTransfer/tree/master/examples) and see if you can get those to work.

Also, please post complete programs, not snippets. Lastly, please use [markdown](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting) when posting code in issues.

jaykorea, 2022-01-12:
I think this is not the pyserialtransfer library bug

I've found out pyserial sometimes occurs input/output error

Do you have any idea of this?

PowerBroker2, 2022-01-15:
Unless you can replicate the issue with a built-in example like those linked in my previous reply, I agree with @jaykorea that this is not a library bug.

closed 2022-01-15 by PowerBroker2

jaykorea, 2022-01-17:
> 이전 답변에 링크된 것과 같은 기본 제공 예제로 문제를 복제할 수 없는 한 나는 이것이 라이브러리 버그가 아니라는 @jaykorea 에 동의합니다.

I think it occurs of Hardware bug or pyserial library bug.

As I add 0.2sec delay, CRC error declines, but It's so slow that apply to other realtime project,,,

thank you powerbroker!