PowerBroker2/pySerialTransfer #48: Issues sending list of integers

knaugh, 2022-01-08:
Hello PowerBroker2,
I am trying to use this library to control some LEDs. I can see that the arduino is getting something from the onboard LED, but I am not getting the correct values. Is there something I am missing in my code?
Python Code

```python
class Hardware:
    def __init__(self):
        
        if comports:
            port_list=list(comports())
            for each in port_list:
                logger.debug(f"port at {each.name}")
                if "Arduino" in each.description:
                    logger.debug(f"arduino found on {each.name}")
                    self.type="arduino"
                    self.link=txfer.SerialTransfer(each.name)
    def open(self):
        self.link.open()
        time.sleep(2)
        self.is_open=True
    def close(self):
        self.link.close()
        self.is_open=False

    def send(self, cmd:Command):
        data=cmd.to_serial() #list of 5 ints
        size=0
        size=self.link.tx_obj(data[0], start_pos=size)
        size=self.link.tx_obj(data[1], start_pos=size)
        size=self.link.tx_obj(data[2], start_pos=size)
        size=self.link.tx_obj(data[3], start_pos=size)
        size=self.link.tx_obj(data[4], start_pos=size)
        logger.debug(size)
        logger.debug(data)
        
        self.link.send(size)
```

And the arduino:
```c++
#include <I2CTransfer.h>
#include <Packet.h>
#include <PacketCRC.h>
#include <SerialTransfer.h>
#include <SPITransfer.h>

#include <Adafruit_NeoPixel.h>

#define LED_PIN 6
#define LED_COUNT 60
SerialTransfer myTransfer;

struct STRUCT{
  int32_t red;
  int32_t green;
  int32_t blue;
  int32_t white;
  int32_t bright;
} command;

Adafruit_NeoPixel strip(LED_COUNT, LED_PIN, NEO_GRBW + NEO_KHZ800);

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  myTransfer.begin(Serial);
  strip.begin();
  strip.show();
  strip.setBrightness(15);
}

void loop() {
  //format RGBWBrightness
  if(myTransfer.available())
  {
    uint16_t recSize=0;
    recSize=myTransfer.rxObj(command.red, recSize);
    recSize=myTransfer.rxObj(command.green, recSize);
    recSize=myTransfer.rxObj(command.blue, recSize);
    recSize=myTransfer.rxObj(command.white, recSize);
    recSize=myTransfer.rxObj(command.bright, recSize);
  }
  
  // put your main code here, to run repeatedly:
  for (int x=0; x<strip.numPixels(); x++){
    strip.setPixelColor(x, command.red, command.green, command.blue, command.white);
  }
  strip.setBrightness(command.bright);
  strip.show();

}
```
Thanks!

PowerBroker2, 2022-01-09:
You'll probably want to use a TTL to USB converter and connect it to another serial port on your arduino and use it for debugging.

First things first: can you get a simple example from the library working? Best to start small with proven examples and then build from there.

Also, you should be able to simplify your code by doing:
```C++
if(myTransfer.available())
  {
    myTransfer.rxObj(command);
  }
```

knaugh, 2022-01-09:
The TTL cable seems like what I will need, I'll go ahead and get one ordered. 
The test code does work, and my code works as well if I wrap the send instructions in a while True loop. 
For now i've gotten it to work by having the arduino send back data when it recieves (and having python loop until it gets that) but that it is slow and not ideal.
I will try to debug the arduino and get back to you. Thanks!

knaugh, 2022-01-14:
Turns out my issue was just that I was using too many LEDs and browning out the atmega. 
Thank you for this library it is a great solution!

closed 2022-01-14 by knaugh