PowerBroker2/pySerialTransfer #49: AttributeError occurs

jaykorea, 2022-01-12:
Hello powerbroker

Nowadays, I'm making an AT command protocol using pyserialTransfer library

However, as I use link.available function, error occurs

Here is the error message and don't know what is the problem of this message

Traceback (most recent call last):
  File "/home/jaycho/carla-simulator/PythonAPI/examples/serial_transmitter.py", line 142, in <module>
    st.transmit_data()
  File "/home/jaycho/carla-simulator/PythonAPI/examples/serial_transmitter.py", line 103, in transmit_data
    if self.link.available():
  File "/home/jaycho/.local/lib/python2.7/site-packages/pySerialTransfer/pySerialTransfer.py", line 511, in available
    recChar = int.from_bytes(self.connection.read(),
AttributeError: type object 'int' has no attribute 'from_bytes'

here is my python code
> def transmit_data(self):
        if self.link.available():
            query_cmd = 0.0
            recSize = 0
            query_cmd = self.link.rx_obj(obj_type='f', start_pos=recSize)
            if query_cmd == '11.0':
                sendsize = 0

                sendsize = self.link.tx_obj(self.vehicle_speed, start_pos=sendsize)
                sendsize = self.link.tx_obj(self.vehicle_throttle, start_pos=sendsize)
                sendsize = self.link.tx_obj(self.vehicle_rpm, start_pos=sendsize)
                sendsize = self.link.tx_obj(self.vehicle_accelX, start_pos=sendsize)
                sendsize = self.link.tx_obj(self.vehicle_accelY, start_pos=sendsize)
                sendsize = self.link.tx_obj(self.vehicle_lat, start_pos=sendsize)
                sendsize = self.link.tx_obj(self.vehicle_lon, start_pos=sendsize)

                self.link.send(sendsize)
                time.sleep(0.01)
            else:
                print("Waiting for Data...")

PowerBroker2, 2022-01-12:
Please post your entire Arduino and Python sketches in tags as described here: [https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting)

jaykorea, 2022-01-12:
> Please post your entire Arduino and Python sketches in tags as described here: https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting

sorry for being idiot... I edited the script now . 
As I change to python3, this error doesn't appear, but still I can not receive "at+qd" string...

PowerBroker2, 2022-01-13:
This is a really odd error. To be honest, I have no idea why Python is complaining about that line. Maybe some more debugging within `rx_obj()` will help...

jaykorea, 2022-01-17:
> 이것은 정말 이상한 오류입니다. 솔직히 말해서, 파이썬이 그 라인에 대해 불평하는 이유를 모르겠습니다. 아마도 더 많은 디버깅 `rx_obj()`이 도움이 될 것입니다 ...

I can not find what's the problem..

so that I simply change python version 2.7 to 3.7 

It works. I think python2 can not find the package module of serial library

Thanks anyway, really big thanks to you

closed 2022-01-17 by PowerBroker2