PowerBroker2/pySerialTransfer #55: Typo in example file and logical error

mickeyvanolst, 2022-04-14:
Hi PowerBroker2,

First off, amazing work, your library is quite impressive!

I'm trying to use it to send a large text file via Python over UART to a Teensy. 

What I noticed is that there is a typo in your example file on this line, the ')' at the end prevents it from compiling.

https://github.com/PowerBroker2/pySerialTransfer/blob/949a7e2197f7bc9c889fcd7f9ad140f79082d444/examples/file/Arduino/rx_file/rx_file.ino#L34

A bigger issue though, is that combined with the Python example file, I can't get any file through. What I noticed through some tests is that the condition `myTransfer.currentPacketID() == 1` never seems to be true.

Just to give you a bit of insight, I'm trying to store the incoming file onto an SD card, below you can see my code.

```
#include <SD.h>
#include <SPI.h>
#include "SerialTransfer.h"

const int fileSize = 2000;
char file[fileSize];
uint16_t fileIndex = 0;
char fileName[10];

SerialTransfer myTransfer;
const int chipSelect = BUILTIN_SDCARD;

void setup()
{
  Serial.begin(115200);
  myTransfer.begin(Serial);

  Serial.print("Initializing SD card...");

  if (!SD.begin(chipSelect)) {
    Serial.println("Card failed, or not present");
    while (1) {
      // No SD card, so don't do anything more - stay stuck here
    }
  }
  Serial.println("card initialized.");

  pinMode(13, OUTPUT);
}

void loop()
{
  if (myTransfer.available())
  {

    // open the file.
    File dataFile = SD.open("datalog.txt", FILE_WRITE);
    if (!myTransfer.currentPacketID())
    {
      myTransfer.rxObj(fileName);
    }
    else if (myTransfer.currentPacketID() == 1)
    {
      myTransfer.rxObj(fileIndex);
      digitalWrite(13, HIGH);

      for (uint8_t i = sizeof(fileIndex); i < myTransfer.bytesRead; i++)
      {
        file[fileIndex] = (char)myTransfer.packet.rxBuff[i];
        if (dataFile) {
          dataFile.print(file[fileIndex]);
        } else {
          // if the file isn't open, pop up an error:
          Serial.println("error opening datalog.txt");
        }
        fileIndex++;
      }
    }
    dataFile.close();
  }
}

```

Any clue what could be going on here?

mickeyvanolst, 2022-04-14:
After looking a bit closer at the examples in the Arduino library, I found that adding a 1 to python line 38. Does make the initial portion of the string occur on the SD card.

`link.send(sendSize, 1)`

mickeyvanolst, 2022-04-18:
Another error in the python script is that the substring isn't created in the right way.

`dataStr = file[fileIndex:fileIndex+dataLen]`

adding +dataLen makes the script work again (along with sending the ID number)

PowerBroker2, 2022-04-19:
Thanks for your eyes on the example code - I made the bug fix. Can you confirm it's working for you?

PowerBroker2, 2022-05-14:
Assuming it's fixed

closed 2022-05-14 by PowerBroker2