PowerBroker2/pySerialTransfer #57: Parsing char from Arduino side

Vounecat, 2022-05-02:
Hello PB2 and a big thank you for your work!
I'm a beginner 
I'm trying to trigger some action in the Arduino Leonardo side when a specific char is received from python.
All the manipulation I tried failed... I probably miss some perspective from this point.

So, here my Python:

```
notesTime = [6079, 7413, 8079, 8746, 10079, ... , 25252]
notesType = [0, 0, 2, 0, 0, .... , 2] ---> notesTime and notesType are long array

try:
    link = txfer.SerialTransfer('COM8', 115200, debug=True)
    link.open()

    start = 0
    step = 10
    counter = 0

    while counter < len(notesTime):
        sendSize = 0

        notesTime_ = notesTime[start:start+step] ---> I split it in order to get under 254 bytes per packet.
        notesType_ = notesType[start:start+step]
        message = "notes"
        
        messageSize = link.tx_obj(message) 
        sendSize += messageSize

        notesTimeSize = link.tx_obj(notesTime_, sendSize) - sendSize
        sendSize += notesTimeSize

        notesTypeSize = link.tx_obj(notesType_, sendSize) - sendSize
        sendSize += notesTypeSize

        link.send(sendSize, packet_id=0) ---> I recently tried to add packet_id information to don't messing up the receiving part.
        counter += len(notesTime_)

        #print('SENT: {} {} {}'.format(message, notesTime_, notesType_))

        while not link.available():
            pass

        rec_message = link.rx_obj(obj_type=type(message), obj_byte_size=messageSize)
        rec_notesTime = link.rx_obj(obj_type=type(notesTime_), obj_byte_size=notesTimeSize, list_format='i', start_pos = messageSize)
        rec_notesType = link.rx_obj(obj_type=type(notesType_), obj_byte_size=notesTypeSize, list_format='i', start_pos = messageSize + notesTimeSize)

        start += step
        print("Recu:\t{} - {} et {}".format(rec_message, rec_notesTime, rec_notesType))

    while True:
        sendSize2 = 0
        message2 = "click"
        messageSize = link.tx_obj(message2)
        sendSize2 += messageSize

        print("Ready to click...")
        link.send(sendSize2, packet_id=0)
        #print('SENT: {}'.format(message2))

        while not link.available():
            pass

        rec_message = link.rx_obj(obj_type=type(message2), obj_byte_size=messageSize)
        print("Recu:\t{}".format(rec_message))

        raise KeyboardInterrupt

except KeyboardInterrupt:
    try:
        link.close()
    except:
        pass

except:
    import traceback
    traceback.print_exc()

    try:
        link.close()
    except:
        pass
```

And my Arduino:

```
#include "SerialTransfer.h"
#include "Keyboard.h"
#include "Mouse.h"

SerialTransfer myTransfer;

int32_t noteTime[10];
int32_t noteType[10];

int32_t noteTimeG[] = {};
int32_t noteTypeG[] = {};

char message[5];

#Stuff I tried....
char notes[] = "notes";
char clickOk[] = "click";

void setup()
{
  Keyboard.begin();
  Mouse.begin();
  Serial.begin(115200);
  myTransfer.begin(Serial, true);
}

void loop()
{
  if(myTransfer.available())
  {
    uint16_t recSize = 0;
    recSize = myTransfer.rxObj(message, recSize);

    if (strcmp(message, clickOk))
    {
      recSize = myTransfer.rxObj(noteTime, recSize);
      recSize = myTransfer.rxObj(noteType, recSize);
  
      memmove(noteTimeG + sizeof(noteTimeG), noteTime, sizeof(noteTime));
      memmove(noteTypeG + sizeof(noteTypeG), noteType, sizeof(noteType));
  
      uint16_t sendSize = 0;
      
      sendSize = myTransfer.txObj(message, sendSize);
      sendSize = myTransfer.txObj(noteTime, sendSize);
      sendSize = myTransfer.txObj(noteType, sendSize);
      
      myTransfer.sendData(sendSize);
     
      delay(50);
    }
    else if (strcmp(message, clickOk))
    {
      uint16_t sendSize2 = 0;
      sendSize2 = myTransfer.txObj(message, sendSize2);
      myTransfer.sendDatum(sendSize2);
      Mouse.click();
    }
  }
}
```

My FTDI card will be coming soon, it should be easier for debugging, but in the meantime any help will be appreciated :)
Maybe my mechanism is completely broken, so feel free to suggest another solution to get what I want.

Bye, 
Vouncat

PowerBroker2, 2022-05-02:
I'm fairly busy these days. Unless it's a quick question, you'll probably get a faster/better answer by asking on the Arduino forum [here](https://forum.arduino.cc/c/using-arduino/interfacing-w-software-on-the-computer/28)

That being said, to better test your code, strip it down to _as simple of code as absolutely possible_ (both tx and rx). Then see what happens. As you mentioned, an FTDI converter makes that process a whole lot easier.

Vounecat, 2022-05-03:
So, as a quick question : how to parse char in arduino after receiving?

```
void loop()
{
  if(myTransfer.available())
  {
    uint16_t recSize = 0;
    recSize = myTransfer.rxObj(message, recSize);

    if (strcmp(message, "test")) ---> Not sure about it
    {
        DO STUFF
```

PowerBroker2, 2022-05-03:
That looks correct. What would be easier IMO would be to use bools, ints, and floats to signify different pieces of data and flags. You can even use a struct or sets of structs to organize everything. Most of my examples (especially on the Arduino side) use structs to show how easy it makes things. In your case, you could do the following:

```C++
#include "SerialTransfer.h"

SerialTransfer myTransfer;
struct __attribute__((__packed__)) STRUCT {
  bool flag;
  float data;
} testStruct;

void setup()
{
  Serial.begin(115200);
  <debug serial port>.begin(115200);
  myTransfer.begin(Serial, true, <debug serial port>);
}

void loop()
{
  if(myTransfer.available())
  {
    myTransfer.rxObj(testStruct);

    if (testStruct.flag) // or !testStruct.flag
    {
        DO STUFF
```

Vounecat, 2022-05-03:
> That looks correct. 

Ah? One guy on Arduino's Forum advised me to use **memcmp**. Anyway, I'll definitely change ma char array to a bool. 

> You can even use a struct

That's a nice suggestion!

Thank you your time !

Vounecat, 2022-05-04:
Hey @PowerBroker2 ,

I want to add/concat each new array X1 I receive to the unique X1G.
What do you suggest me?

I couldI use some for loop with counter...but the memcpy/memmove aka one line solution seems nice.

PowerBroker2, 2022-05-04:
I'm not sure on this one; I would try asking on the Arduino forum

closed 2022-05-11 by PowerBroker2