PowerBroker2/pySerialTransfer #61: How to transmit large array of floats fast and reliably?

Haoyu-R, 2022-08-26:
Hi PowerBroker,

Thank you for your great library. I'm using it to transmit a big array of floats with more than 1000 elements between the laptop and an Arduino Nano BLE board (bi-direction).

I successfully make the transmission work by formatting each float as a string and sending each element one by one. However, it was a bit slow when sending from the laptop to the board (more than 15 seconds).

Now, I chunk the array to fit the packet size of 254 bytes and try to send 63 floats each time. In this way, I try to minimize sending frequency with the hope to increase speed.

However, the transmission is not getting unreliable now. Some packets are not received by the board and got lost.

Could you please help me?

**Arduino code:**

```
#include <Arduino.h>
#include "SerialTransfer.h"

#define weightsSize 1153 // Unit is byte
#define transferSize 254
#define floatSize 4

// ---------------------------------- Serial transfer parameter ---------------------------------------

SerialTransfer myTransfer;
bool train = false;
bool receive = true;  // First, receive weights from the server
bool startReceive = false;
bool send = false;
int weightIndex = 0;
char numString[11];
char command[6];
char floatWeight[11];
int receiveFloatsPerRound = transferSize / floatSize;
int receiveRound = int(weightsSize / receiveFloatsPerRound) + 1; 
int currentReceiveRound = 0;
int lastRoundFloatsNumber = weightsSize - (receiveRound - 1) * receiveFloatsPerRound;
// float floatsHolder[int(transferSize / floatSize)]; 
float flatWeights[1153] = {0};
// float myFloat = 0;
// -------------------------------------------------------------------------------------------

void setup() {
  Serial.begin(115200); //115200 baud rate (If necessary, change in the serial monitor)
  myTransfer.begin(Serial);
  delay(2000);
}

void loop() {

  
    // Send the weights
    if(send){

      myTransfer.sendDatum("start");
      delay(500);

      for(int i = 0; i<weightsSize; i++){
        snprintf_P(floatWeight, 11, "%.6f", flatWeights[i]);
        uint16_t sendSize = myTransfer.txObj(floatWeight[0], 0, 11);
        myTransfer.sendData(sendSize, 1);
        delay(1);
      }

      myTransfer.sendDatum("end");
      send = false;
      receive = true;
      weightIndex = 0;
    }

    if(receive){
      if(myTransfer.available()){

        if(!myTransfer.currentPacketID()){
          myTransfer.rxObj(command);

          if(strstr(command, "end")){
            weightIndex = 0;
            train = true;
            startReceive = false;
            receive = false;
            currentReceiveRound = 0;
          }else if(strstr(command, "start")){
            startReceive = true;
          }
        }
        
        if(myTransfer.currentPacketID() == 1 and startReceive){
            float floatsHolder[int(transferSize / floatSize)]; 
            myTransfer.rxObj(floatsHolder);
            if(currentReceiveRound == receiveRound-1){ // The last round
              for(int j=0; j < lastRoundFloatsNumber; j++){
                flatWeights[currentReceiveRound*receiveFloatsPerRound + j] = floatsHolder[j];
              }
            }else{
              for(int j=0; j < receiveFloatsPerRound; j++){
                flatWeights[currentReceiveRound*receiveFloatsPerRound + j] = floatsHolder[j];
              }
            }
            currentReceiveRound++;
        } 
      }
    }

    if(train)    
    {
      train = false;
      send = true;
      for(int i=0;i<weightsSize;i++){
        flatWeights[i] = flatWeights[i] * 2;
      }
    }
}

```

**Python code:**
```
import random
import time
from pySerialTransfer import pySerialTransfer as txfer
import numpy as np

start_sending = True
com_name = 'COM10'
weights = [np.float16(random.random()) for w in range(1153)] // Here an array with a size of 1153

receiveFloatsPerRound = int(254 // 4)
receiveRound = len(weights) // receiveFloatsPerRound + 1
lastRoundFloatsNumber = len(weights) - (receiveRound - 1) * receiveFloatsPerRound

try:
    start_receive = False
    # Open a COM given its name
    link = txfer.SerialTransfer(com_name)
    link.open()
    time.sleep(2)

    t = 0
    print("Open link...")
    while True:
        if link.available() and not start_sending:
            if not link.idByte:
                state_indicator = link.rx_obj(str, obj_byte_size=link.bytesRead)
                # Central is notified for the start of receiving weights from the device
                if "start" in state_indicator:
                    print('Elapsed time between two rounds (with device) is {:.2f} seconds...'.format(time.time() - t))
                    t = time.time()
                    print("Start receiving...")
                    start_receive = True
                # Central is notified for the end of receiving weights from the device
                if "end" in state_indicator and start_receive:
                    print("End receiving...")
                    print('Elapsed time for receiving is {:.2f} seconds...'.format(time.time() - t))
                    print("Weights received...: ")
                    print(weights)
                    print("length: {}".format(len(weights)))
                    start_receive = False
                    if len(weights) == 1153:
                        print("Devide weights by 2...")
                        weights = [w / 2 for w in weights]
                    start_sending = True
            else:
                if start_receive:
                    # Receive the weights as a string one by one
                    next_content = link.rx_obj(str, start_pos=0, obj_byte_size=link.bytesRead)
                    weights.append(float(next_content.replace('\x00', '')))

        if start_sending:

            # Notify the device for receiving incoming weights
            link.send(link.tx_obj("start"))
            time.sleep(2)
            t = time.time()
            print("Start sending...")
            for i in range(receiveRound):
                if i == receiveRound - 1:
                    list_size = link.tx_obj(weights[i*receiveFloatsPerRound:i*receiveFloatsPerRound+lastRoundFloatsNumber])
                    print(weights[i*receiveFloatsPerRound:i*receiveFloatsPerRound+lastRoundFloatsNumber])
                    link.send(list_size, packet_id=1)
                else:
                    print(weights[i*receiveFloatsPerRound:(i+1)*receiveFloatsPerRound])
                    list_size = link.tx_obj(weights[i*receiveFloatsPerRound:(i+1)*receiveFloatsPerRound])
                    link.send(list_size, packet_id=1)
                time.sleep(0.1)
            weights.clear()
            link.send(link.tx_obj("end"))
            print("End sending...")
            print('Elapsed time for sending is {:.2f} seconds...'.format(time.time() - t))
            t = time.time()
            # After send weights to the device, toggle flag
            start_sending = False

        if link.status < 0:
            if link.status == txfer.CRC_ERROR:
                print('ERROR: CRC_ERROR')
            elif link.status == txfer.PAYLOAD_ERROR:
                print('ERROR: PAYLOAD_ERROR')
            elif link.status == txfer.STOP_BYTE_ERROR:
                print('ERROR: STOP_BYTE_ERROR')
            else:
                print('ERROR: {}'.format(link.status))

except KeyboardInterrupt:
    print("KeyboardInterrupt")
    link.close()
except:
    import traceback

    traceback.print_exc()
    print("Error")
    try:
        link.close()
    except:
        pass

```

PowerBroker2, 2022-08-27:
I've had success in adding a 10ms delay between packet sends. Maybe it will help here, too?

Haoyu-R, 2022-08-27:
Hi PB2, Thanks a lot.

My final workaround is to format several floats into a string at the computer side each time and send the string to the board. On the board side, the string is parsed and each float is put back into an array.

The observation is that, with a packet length of 254, transmission is unreliable. Data still get lost no matter how much delay is introduced.

Finally, I reduce each packet to a size of 128. Now, everything works. ;-)

closed 2022-08-27 by PowerBroker2