PowerBroker2/pySerialTransfer #64: Sending a Boolean

jmdelahanty, 2022-10-07:
Hello glorious PowerBroker!

I'm trying to send along a boolean value in a struct but I'm not sure how to do it with pySerialTransfer! Any advice?

PowerBroker2, 2022-10-07:
Hey Jeremy!

Bools are 1 byte objects both in [Arduino/C++](https://sites.uclouvain.be/SystInfo/usr/include/stdbool.h.html) and [Python](https://docs.python.org/3/library/struct.html#format-characters), so they should be compatible automatically. Bools are also [automatically supported in pySerialTransfer](https://github.com/PowerBroker2/pySerialTransfer/blob/782c4ab712a530eea797783e01a18941e5227cc3/pySerialTransfer/pySerialTransfer.py#L267-L268).

jmdelahanty, 2022-10-07:
You're the best! I'll close this for now. If I run into something weird when I'm putting it into the code I'll just comment here.

closed 2022-10-07 by jmdelahanty

jmdelahanty, 2022-10-07:
To double check, this means I don't need to do any `val_type_overrides` if what's fed into the `tx_buffer` is type `bool`? See [this line](https://github.com/Tyelab/bruker_control/blob/6c496a8195304d14a6c6004179925611de909344/main/serialtransfer_utils.py#L243) for where I'm doing it.

So I could just add this to the transfer:

`metaData_size = link.tx_obj(arduino_metadata['lickContingency'], metaData_size)`

`lickContingency` is a Boolean value (literally `True` or `False`)

PowerBroker2, 2022-10-07:
`metaData_size = link.tx_obj(arduino_metadata['lickContingency'], metaData_size)` should be fine because Python will automatically create a value type format for the transmission if the payload object is of a built-in Python type (i.e. bool):

tx_obj():
https://github.com/PowerBroker2/pySerialTransfer/blob/782c4ab712a530eea797783e01a18941e5227cc3/pySerialTransfer/pySerialTransfer.py#L253-L274

I'd say give it a shot and see if it works

jmdelahanty, 2023-01-04:
This totally works by the way!