PowerBroker2/pySerialTransfer #65: Is there a maximum variable size for "x" when using myTransfer.sendDatum("x")?

nolan4, 2022-10-24:
Hi, I'm trying to send a structure from a microcontroller to my computer. The structure has 7 fields and each field is a "packetSize" element array. Here is the structure initialization on the microcontroller:

const int packetSize = 5;
struct STRUCT {
  int ax[packetSize];
  int ay[packetSize];
  int az[packetSize];
  int gx[packetSize];
  int gy[packetSize];
  int gz[packetSize];
  int ts[packetSize];
} sendStruct;

The python code on my computer looks like this:
data = link.rx_obj(obj_type=list, obj_byte_size=4*5*7, list_format='i')

In the above python code, 4 is the bytes per number, 5 is the "packetSize," and 7 is the number of fields in my structure. I'm trying to maximize the "packetSize" here. When I upload my arduino code, I can put any number for packetSize, but if I want to receive the data through python the largest packetSize I can use is 9. A packetSize of 9 or less works perfectly, but anything greater than 9 gives the following error: 

//
arr = array(list_format, buff)
ValueError: bytes length not a multiple of item size
//

Is there a way I can use a packetSize of more than 9 or is this the maximum amount of data that can be sent at once?

nolan4, 2022-10-24:
Thanks for your fast reply! Where would I specify int32_t? Here is what I now have:
const int packetSize = 20;
struct `__attribute__((packed))` STRUCT {
  int ax[packetSize];
  int ay[packetSize];
  int az[packetSize];
  int gx[packetSize];
  int gy[packetSize];
  int gz[packetSize];
  int ts[packetSize];
} sendStruct;

`data = link.rx_obj(obj_type=list, obj_byte_size= 4 *  20 *  7, list_format='h')`

I do get data transferred this way, but the formatting is odd (separated by -1s and 0s?) and I am not receiving all of the data. For a packetSize of 20, I should get 140 values. I only receive 64 values (excluding the -1s and 0s) and they are shown here:

[-5152, -1, -5152, -1, -5184, -1, -5200, -1, -5248, -1, -5224, -1, -5232, -1, -5232, -1, -5328, -1, -5192, -1, -5312, -1, -5192, -1, -5192, -1, -5096, -1, -5464, -1, -5464, -1, -5464, -1, -5232, -1, -5184, -1, -5232, -1, 14248, 0, 14248, 0, 14376, 0, 14344, 0, 14392, 0, 14592, 0, 14496, 0, 14496, 0, 14416, 0, 14352, 0, 14464, 0, 14520, 0, 14520, 0, 14496, 0, 14488, 0, 14488, 0, 14616, 0, 14664, 0, 14552, 0, 14576, 0, -4520, -1, -4520, -1, -4432, -1, -4432, -1, -4584, -1, -4648, -1, -4648, -1, -4648, -1, -4504, -1, -4512, -1, -4600, -1, -4736, -1, -4736, -1, -4624, -1, -4408, -1, -4408, -1, -4448, -1, -4536, -1, -4592, -1, -4504, -1, 43, 0, 135, 0, -8, -1, -31]

PowerBroker2, 2022-10-24:
The values you're trying to transfer are most likely from an IMU and probably should be floats, which is actually easier for the libraries to handle automatically. I'd suggest switching to floats and seeing if that doesn't fix your issue.

PowerBroker2, 2022-10-24:
> Where would I specify int32_t?

In the Arduino code inside your struct def like this:

```C++
struct __attribute__((packed)) STRUCT {
int32_t ax[packetSize];
int32_t ay[packetSize];
int32_t az[packetSize];
int32_t gx[packetSize];
int32_t gy[packetSize];
int32_t gz[packetSize];
int32_t ts[packetSize];
} sendStruct;
```

nolan4, 2022-10-25:
Yes you're right about the IMU data. To record faster I just store the raw int16_t values coming from the accelerometer and gyroscope. The first 6 values come through correctly when I use int16_t in the struct def. The only issue is the timestamp which is the difference between two unsigned longs. Is there a correct way to determine the obj_byte_size for this struct? The code/output I have now for a packetSize of 10 is:

const int `packetSize` = 10;
struct `__attribute__((packed))` STRUCT {
  int16_t ax[packetSize];
  int16_t ay[packetSize];
  int16_t az[packetSize];
  int16_t gx[packetSize];
  int16_t gy[packetSize];
  int16_t gz[packetSize];
  unsigned long ts[packetSize];
} imuStruct;

`y = link.rx_obj(obj_type=list, obj_byte_size=(2*packetSize*6 + 4*packetSize), list_format='h')`

acc_x: [-5168, -5432, -5328, -5328, -5136, -5216, -5400, -5496, -5496, -5264], 
acc_y: [14480, 14520, 14536, 14536, 14456, 14648, 14576, 14352, 14352, 14496], 
acc_z: [-4600, -4616, -4456, -4456, -4608, -4776, -4552, -4472, -4472, -4496], 
gyr_x: [137, 124, -61, -54, -13, 124, -66, -3, -67, 135], 
gyr_y: [19, 30, 88, 130, -387, -104, -103, -34, -120, 94], 
gyr_z: [-216, 170, -338, 262, -81, -65, -213, -17, -173, -4], 
ts: [3121, 0, 3299, 0, 3471, 0, 3634, 0, 3797, 0]]

PowerBroker2, 2022-10-25:
In that case, you can use `list_format='h'` for the 16-bit values and `list_format='i'` for the long values. Take a look at the pySerialTransfer readme example to see how to parse multiple objects from a single packet

nolan4, 2022-10-25:
Ok based on the readme code, this is what I am now doing. 

Here is the arduino code:

const int packetSize = 20;
struct `__attribute__((packed))` STRUCT {
  int16_t ax[packetSize];
  int16_t ay[packetSize];
  int16_t az[packetSize];
  int16_t gx[packetSize];
  int16_t gy[packetSize];
  int16_t gz[packetSize];
  unsigned long ts[packetSize];
} imuStruct;

Here is the python code (based on the readme):

```
packetSize = 20

acc_size = 2*packetSize*3     # bytes required for 20*3 int16_t values
gyr_size = 2*packetSize*3      # bytes required for 20*3 int16_t values
ts_size = 4*packetSize           # bytes required for 20*3 unsigned long values

a = link.rx_obj(obj_type=list, obj_byte_size=acc_size, start_pos=0, list_format='h')
g = link.rx_obj(obj_type=list, obj_byte_size=gyr_size, start_pos=acc_size, list_format='h')
ts = link.rx_obj(obj_type=list, obj_byte_size=ts_size, start_pos=(acc_size + gyr_size), list_format='i')
```

Here is the error:

```
ts = link.rx_obj(obj_type=list, obj_byte_size=ts_size, start_pos=(acc_size + gyr_size), list_format='i')
  File "/home/nolan/.local/lib/python3.8/site-packages/pySerialTransfer/pySerialTransfer.py", line 355, in rx_obj
    arr = array(list_format, buff)
ValueError: bytes length not a multiple of item size
```

nolan4, 2022-10-25:
It works correctly for a packetSize of 15. Here is the output:

a: [-4024, -4080, -4120, -4144, -4144, -4264, -4096, -4096, -3832, -3936, -4008, -4008, -3976, -3976, -4008, 15648, 15560, 15536, 15648, 15552, 15456, 15504, 15504, 15720, 15600, 15424, 15424, 15512, 15544, 15624, -2016, -1968, -2192, -2144, -1976, -2120, -2152, -2152, -2144, -2064, -1976, -1976, -2008, -2072, -2000] 

g: [-80, -86, -421, -151, 80, 66, 379, -202, 53, -58, 344, -76, 220, -129, 216, -239, 352, -232, 368, -79, 18, -341, -86, -20, -420, 145, -140, 491, -136, -4, -153, -158, -157, 265, -143, 385, -297, 74, -296, 164, -381, 188, -358, 111, -49] 

ts: [1020, 1201, 1369, 1644, 1807, 1970, 2133, 2296, 2459, 2649, 2812, 2975, 3139, 3302, 3465]

PowerBroker2, 2022-10-25:
(2*20*3) + (2*20*3) + (4*20*3) = 480

You can't stuff more than 254 bytes per packet

nolan4, 2022-10-25:
Ok yes that is the issue. Is there a way I can experiment with increasing this limit? Thanks for your help!

PowerBroker2, 2022-10-26:
The max packet size is limited due to using a 1 byte [COBS overhead field](https://en.wikipedia.org/wiki/Consistent_Overhead_Byte_Stuffing). If you need to send a multi-packet transmission, look at the [file transfer examples](https://github.com/PowerBroker2/pySerialTransfer/tree/master/examples/file)

nolan4, 2022-10-26:
Sounds good! Thanks again.

closed 2022-10-26 by nolan4