PowerBroker2/pySerialTransfer #66: Serial port doesn’t stay connected

AleMontii, 2022-11-14:
Hi first of all thanks for your work on this amazing library!
I'm using it to transmit a bunch of data 25 times a second from a Mac to a Teensy (basically a more powerful Arduino) to display synced animations on led strips. Everything works perfectly except when sometimes... it doesn't. It randomly disconnects, sometimes more frequently, sometimes never, other times just once and restarting fixes it, and I can't get over what may be the cause. With some googling I suspect there might be some Getty process taking over my program and therefore disconnecting the serial port, but couldn't find any solution. I'm posting this here in the hope that maybe you, or some other user, encountered such issues as well and could help.
Thanks!

```
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/serial/serialposix.py", line 621, in write
    n = os.write(self.fd, d)
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pySerialTransfer/pySerialTransfer.py", line 481, in send
    self.connection.write(stack)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/serial/serialposix.py", line 655, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 6] Device not configured
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/serial/serialposix.py", line 621, in write
    n = os.write(self.fd, d)
OSError: [Errno 6] Device not configured
```

PowerBroker2, 2022-11-14:
Never had this problem before. Maybe these links will help?:

[https://stackoverflow.com/a/31530670/9860973](https://stackoverflow.com/a/31530670/9860973)
[https://esp32.com/viewtopic.php?t=2595](https://esp32.com/viewtopic.php?t=2595) <- See where multiple users said changing the USB cable fixed the error

AleMontii, 2022-11-18:
> https://stackoverflow.com/a/31530670/9860973 https://esp32.com/viewtopic.php?t=2595 **<- See where multiple users said changing the USB cable fixed the error**

Aaand yes that fixed the problem. I can't explain myself how I've never thought to check the cable for such a long time, so thank you for finally solving this issue that has been bothering me for way too long. Thanks again!

closed 2022-11-18 by AleMontii