PowerBroker2/pySerialTransfer #67: Send encoder data

w1ngedshadow, 2022-12-31:
Hello, I am building mobile robot based on nvidia jetson nano and arduino. I have two motor encoders connected to arduino so I need to send data to jetson nano (python) via uart. What is the best and right way to do it with this library? Approximate range of encoder values is 0 - 1000000. I will appreciate your help.

PowerBroker2, 2022-12-31:
A good place to start is with these examples:

[Arduino TX Data (C++)](https://github.com/PowerBroker2/pySerialTransfer/blob/master/examples/data/Arduino/tx_data/tx_data.ino)
[Nano RX Data (Python)](https://github.com/PowerBroker2/pySerialTransfer/blob/master/examples/data/Python/rx_data.py)

You can find more examples [here](https://github.com/PowerBroker2/pySerialTransfer/tree/master/examples), too.

If you have questions on implementation, the doc strings should be able to answer most of them. For convenience, here's the pySerialTransfer doc strings for the `tx_obj()` and `rx_obj()` functions:

https://github.com/PowerBroker2/pySerialTransfer/blob/782c4ab712a530eea797783e01a18941e5227cc3/pySerialTransfer/pySerialTransfer.py#L230-L247

https://github.com/PowerBroker2/pySerialTransfer/blob/782c4ab712a530eea797783e01a18941e5227cc3/pySerialTransfer/pySerialTransfer.py#L309-L333

w1ngedshadow, 2023-01-01:
I forget to mention that I use not an usual Arduino like Nano but Arduino Zero. So I use SerialUSB instead of usual Serial. With examples given upper I have following error on python side: `b'$'0.0 | @h` is my output. I think its something simular to #51.
What is wrong ?)

PowerBroker2, 2023-01-01:
Can you post your exact Python and Arduino sketches (with proper markdown) _and_ the full error traceback?

w1ngedshadow, 2023-01-02:
Maybe I am mistaken in description, there arent any errors, I have wrong output on python side: ` b'$'0.0 | @h` instead of `b'$'4.5 | hello`

**Nano code:**

```
from time import sleep
from pySerialTransfer import pySerialTransfer as txfer

class struct(object):
    z = ''
    y = 0.0

arr = ''

if __name__ == '__main__':
    try:
        testStruct = struct
        link = txfer.SerialTransfer('/dev/ttyACM0')
        
        link.open()
        sleep(5)
    
        while True:
            if link.available():
                recSize = 0
                
                testStruct.z = link.rx_obj(obj_type='c', start_pos=recSize)
                recSize += txfer.STRUCT_FORMAT_LENGTHS['c']
                
                testStruct.y = link.rx_obj(obj_type='f', start_pos=recSize)
                recSize += txfer.STRUCT_FORMAT_LENGTHS['f']
                
                arr = link.rx_obj(obj_type=str,
                                  start_pos=recSize,
                                  obj_byte_size=5)
                recSize += len(arr)
                
                print('{}{} | {}'.format(testStruct.z, testStruct.y, arr))
                
            elif link.status < 0:
                if link.status == txfer.CRC_ERROR:
                    print('ERROR: CRC_ERROR')
                elif link.status == txfer.PAYLOAD_ERROR:
                    print('ERROR: PAYLOAD_ERROR')
                elif link.status == txfer.STOP_BYTE_ERROR:
                    print('ERROR: STOP_BYTE_ERROR')
                else:
                    print('ERROR: {}'.format(link.status))
                
        
    except KeyboardInterrupt:
        link.close()
```

**Arduino code:**

```
#include "SerialTransfer.h"

SerialTransfer myTransfer;

struct __attribute__((packed)) STRUCT {
  char z;
  double y;
} testStruct;

char arr[] = "hello";

void setup()
{
  SerialUSB.begin(115200);
  myTransfer.begin(SerialUSB);

  testStruct.z = '$';
  testStruct.y = 4.5;
}

void loop()
{
  // use this variable to keep track of how many
  // bytes we're stuffing in the transmit buffer
  uint16_t sendSize = 0;

  ///////////////////////////////////////// Stuff buffer with struct
  sendSize = myTransfer.txObj(testStruct, sendSize);

  ///////////////////////////////////////// Stuff buffer with array
  sendSize = myTransfer.txObj(arr, sendSize, strlen(arr));

  ///////////////////////////////////////// Send buffer
  myTransfer.sendData(sendSize);
  delay(500);
}
```

PowerBroker2, 2023-01-02:
Ok, I see. I don't know why it wouldn't work or why you wouldn't get errors if corrupted packets were received. I would suggest comparing the bytes sent by the Arduino and those received by the Nano. If both match, the issue is somehow with the parsing. You might get faster/better help using the Arduino forum

closed 2023-01-03 by w1ngedshadow