PowerBroker2/pySerialTransfer #68: Can we use this library on Raspberry Pi for communication through GPIO?

v-r-a, 2023-04-01:
Hello,

I am a PhD student working on biped robots.

Thank you for this wonderful library.

I have been using SerialTransfer on Arduino Mega, Uno, Nano, etc., for two-way communication between Arduinos. I am using an RS-485 half-duplex multi-drop setup for my work. Example: Arduino Mega is the master and requests sensor data from four Arduino Nano boards in a loop.

I want to add Raspberry Pi to my RS-485 bus. I am using an [RS-485 shield](https://www.waveshare.com/2-ch-rs485-hat.htm) (uses GPIO pins) for the same.  Can I use SerialTransfer/ pySerialTransfer for this communication?

PowerBroker2, 2023-04-01:
I've used SerialTransfer and pySerialTransfer to send data between an Arduino and RPi, but that was via USB. However, I'm sure using GPIO pins works just as good.

v-r-a, 2023-04-01:
I just read your suggestion on USB to RS485 converter in #44. 

I am yet to do trials with the shield I have. Hope it works.

Thank you!

closed 2023-04-01 by v-r-a