PowerBroker2/pySerialTransfer #70: Data array sometimes gets corrupted if some values are present

AleMontii, 2023-11-23:
I'm using the library to transmit unsigned 8-bit integers to a Teensy 4.1 from a Mac. The data is used to drive 24 channels of 256 RGB addressable LEDs each, so each frame is made up of 24 * 256 * 3 values between 0 and 255. This needs to run at 25fps in sync with what's playing on the computer. The data gets created, transmitted, and pushed to the LEDs in ~20ms so it works completely fine in that regard. Thanks for your great work on the library! Sadly I'm encountering a weird annoying bug that I'm not able to fix.

I've noticed this issue because of some flashing on the LEDs: in some cases when some value in the array is `126` the data gets corrupted during transmission. I haven't obviously tried all the possible combinations, but by playing around with it a bit I found some cases that don't work, and they all involve the number 126. I've stripped my code to the bare minimum, which is provided below, and did some tests:

```
126	0	126
126	0	126
126	0	126
126	0	126
```
becomes
```
126	0	1
2	0	1
2	0	1
2	0	0
```
but
```
0	126	0
0	126	0
0	126	0
0	126	0
```
and
```
0	126	0
0	126	0
126	0	126
126	0	126
```
both transfer successfully.

I could add many more examples but I think it's easier if you try playing with the code yourself and see if you can reproduce the error. Change the `data` array as well as `SIZE` and `PACKETAMOUNT`. On the full program, I'm working with `SIZE = 64` and `PACKETAMOUNT = 96` but it doesn't seem to change the behaviour much. Note that the error doesn't always follow a predictable pattern.

For example with `SIZE = 64` and `PACKETAMOUNT = 1` an array full of `126`s becomes:
```
126	1	1
1	1	1
...
1	1	1
126	126	126
...
126	126	126
```

### Python code:
```python
from pySerialTransfer import pySerialTransfer as txfer
from random import randint
import struct, sys
import numpy as np

link = txfer.SerialTransfer("/dev/cu.usbmodem103259301")
link.open()

SIZE = 4
PACKETLENGTH = SIZE*3
PACKETAMOUNT = 2

data = [126,0,126]*SIZE*PACKETAMOUNT

if len(data) < PACKETLENGTH*PACKETAMOUNT:
    sys.exit("len(data) must be equal to PACKETLENGTH*PACKETAMOUNT")
    
print(np.array(data).reshape((SIZE*PACKETAMOUNT, 3)))

for i in range(PACKETAMOUNT):
    if i == PACKETAMOUNT - 1:
        ID = 1 #last packet
    else:
        ID = 0

    #split data into packets
    packets = [data[i:i+PACKETLENGTH] for i in range(0, len(data), PACKETLENGTH)]

    sendSize = link.tx_struct_obj(struct.pack(f">{len(packets[i])}B", *packets[i]))
    link.send(sendSize, packet_id = ID)
```
_I'm using `sendSize = link.tx_struct_obj(struct.pack(f">{len(packets[i])}B", *packets[i]))` because is much faster than iterating over every value and using `val_type_override='B'`, but the issues are the same with either approach._

### Arduino code:
```c++
#include "SerialTransfer.h"

SerialTransfer Transfer;

#define SIZE 4
#define PACKETLENGTH SIZE*3
#define PACKETAMOUNT 2

uint8_t databuffer[PACKETLENGTH]; 
uint8_t ledsbuffer[PACKETLENGTH*PACKETAMOUNT];
int packetnumber = 0;

void setup() {
  Serial.begin(115200);
  Transfer.begin(Serial);
}

void loop() {
  if (Transfer.available()){ 
    int recSize = 0;
    recSize = Transfer.rxObj(databuffer, recSize);
    
    if (recSize != 0){
      for (int i = 0; i < PACKETLENGTH; i++){
        ledsbuffer[packetnumber * PACKETLENGTH + i] = databuffer[i];
      }
      packetnumber++;
    }
    
    if (Transfer.currentPacketID() == 1) { //last packet
      packetnumber = 0;
      for (int i = 0; i < PACKETLENGTH*PACKETAMOUNT; i += 3) {
        Serial.print(ledsbuffer[i]);
        Serial.print("\t");
        Serial.print(ledsbuffer[i+1]);
        Serial.print("\t");
        Serial.print(ledsbuffer[i+2]);
        Serial.println("");
      }
      Serial.println("-------------------");
    }
  }
}
```

Thanks a lot if you could help!

PowerBroker2, 2023-11-25:
It probably has something to do with 126 being the same as the packet start byte:

https://github.com/PowerBroker2/SerialTransfer/blob/c596233371af2735c1a20be8f70f5e5ce30bb278/src/Packet.h#L30

However, consistent overhead byte stuffing (COBS) has been implemented to prevent this exact issue - perhaps it isn't working as expected...

I can't test this right now, but would you mind poking around and seeing what value the COBS byte is for various packets (with and without 126)?

AleMontii, 2023-11-28:
Sorry for taking long to reply, I didn't have the hardware on hand these past days. I had no idea what COBS was until now, so I also looked into it a bit and did some tests by adding `debugPort->println(recChar);` after [this](https://github.com/PowerBroker2/SerialTransfer/blob/c596233371af2735c1a20be8f70f5e5ce30bb278/src/Packet.cpp#L137) line. I understood the general idea of how it works, but I have very little knowledge on this stuff and I couldn't figure out myself where and why it goes wrong.

These are a few arrays that transfer as expected:

Data:  `1 2 3 4 5 6`
Packet: `126 1 255 6 1 2 3 4 5 6 103 129`

Data:  `126 0 0 0 0 0`
Packet: `126 1 0 6 126 0 0 0 0 0 6 129`

Data:  `0, 126, 126, 0, 0, 0`
Packet: `126 1 1 6 0 1 0 0 0 0 229 129`

Data:  `0 0 126 126 0 0`
Packet: `126 1 2 6 0 0 1 0 0 0 143 129`

And a few that don't:
Data sent:  `126 0 126 0 126 0`
Packet: `126 1 0 6 126 0 2 0 0 0 131 129`
Data received:  `126 0 2 0 0 0`

Data sent: `126 126 126 0 0 0`
Packet: `126 1 0 6 126 1 0 0 0 0 227 129`
Data received:  `126 1 0 0 0 0`

Data sent: `126 0 0 126 0 0`
Packet: `126 1 0 6 126 0 0 0 0 0 6 129`
Data received:  `126 0 0 0 0 0`

Data sent: `126 126 126 126 126 126`
Packet: `126 1 0 6 126 1 1 1 1 0 36 129`
Data received:  `126 1 1 1 1 0`

Let me know if you need anything else and thanks a lot for the help!

PowerBroker2, 2023-11-29:
> I had no idea what COBS was until now, so I also looked into it a bit and did some tests

You've probably already seen it, but [this wiki article has a pretty good explanation of how it all works.](https://en.wikipedia.org/wiki/Consistent_Overhead_Byte_Stuffing)

Looks like the COBS overhead byte (the third byte in each packet) is being set properly. However, it seems that not all instances of 126 within the payload are being replaced correctly. I think the issue may lie in a couple of places:

On the TX side:

https://github.com/PowerBroker2/SerialTransfer/blob/c596233371af2735c1a20be8f70f5e5ce30bb278/src/Packet.cpp#L361-L390

and on the RX side:

https://github.com/PowerBroker2/SerialTransfer/blob/c596233371af2735c1a20be8f70f5e5ce30bb278/src/Packet.cpp#L393-L421

I don't immediately see the issue(s), but I will look into it further...

PowerBroker2, 2023-11-29:
I think I found [the issue](https://github.com/PowerBroker2/pySerialTransfer/commit/b9b05997c18f715f40874a6c33ae5182c015b7df) on the Python side. Please retest with the current version of the master branch.

AleMontii, 2023-11-29:
Yes, that fixed it! Thank you

commit b9b05997 (2023-11-29): Attempt to fix #70
closed 2023-11-29 by AleMontii