PowerBroker2/pySerialTransfer #74: Enhancement Proposal: Modernizing pySerialTransfer Python Client for Improved Code Quality and Practices

robberwick, 2024-03-31:
Firstly, I want to express my gratitude for the continuous maintenance of pySerialTransfer (and serialTransfer). Over the years, I've extensively utilized them and even contributed in the past. I really appreciate that they provide a straightforward solution for packetized data transfer.

Recently, I've been contemplating whether the Python client could benefit from some modernization in terms of code style and practices. I believe updating it to leverage modern Python language features and best practices could enhance its usability and maintainability.

Areas for Improvement:

Language Additions: 
Incorporating features like type hinting, enums, and utilizing match instead of multiple if...else blocks.
Code Formatting: 
Enforcing PEP8 compliance throughout the codebase, utilizing tools like Black or Darker for consistent code styling, and optimizing imports with isort.
Packaging: 
Transitioning to using pyproject.toml for packaging instead of the older setup.py.
Testing: 
Adding pytest-based unit tests to ensure the project's functionality and provide confidence to potential users.

This inquiry isn't a formal issue but rather an exploration of the project's roadmap and the community's interest in such development work. I'm willing to contribute by submitting pull requests, but I wanted to gauge the community's receptiveness and identify the lowest supported version of Python (presumably at least 3.8, considering 3.7's EOL) for tailoring the changes accordingly.

Your feedback and guidance on this proposal would be highly appreciated. Thank you for considering these potential enhancements to pySerialTransfer.

PowerBroker2, 2024-04-02:
Thank you for the proposal and kind words - I'm glad you find the software useful, and I'm super appreciative of your contributions to the projects.

Here's my 2 cents:

> Incorporating features like type hinting, enums, and utilizing match instead of multiple if...else blocks.

Yes, this would be good

> Code Formatting:
> Enforcing PEP8 compliance throughout the codebase, utilizing tools like Black or Darker for consistent code styling, and optimizing imports with isort.

Maybe, but I'm very particular in how I use whitespace - nobody will change that in the codebase no matter what PEP8 says.

> Packaging:
> Transitioning to using pyproject.toml for packaging instead of the older setup.py.

Unfortunately, I'm familiar with setup.py. Maybe if using pyproject.toml isn't to hard to learn/use I could transition this and my other python packages.

> Testing:
> Adding pytest-based unit tests to ensure the project's functionality and provide confidence to potential users.

This would be cool. I would say it's a lower priority, but it certainly would be a good way to make the lib more robust.

robberwick, 2024-04-02:
Ok, so it sounds like there's interest in at least some of those things then, so that's great :+1:

I think most if not all of those areas could be addressed separately and incrementally in individual focused PRs, rather than one big bang, so you should be able to pick and choose which ones you want to incorporate into the project on a case by case basis. 

I'll raise issues and PRs for the various things as I get to them.

Thanks again for all your efforts on the project :+1: